nickg/nvc #1066: [Low Priority] Operator ambiguity re: recent updates to NVC master branch

a-panella, 2024-11-11:
Was wondering if anything significant has changed re: operators and how ambiguity has changed. I am trying to upgrade my regression tests to use NVC master, but there appears to be a change since `a99c2cc9371e750d9c85e047134335511bf56fc0` that causes `&` operator to be more ambigious in master than it was in `a99c2cc9371e750d9c85e047134335511bf56fc0`.

I tend to use a pkg that declares subtypes for `std_logic` and `std_logic_vector` as `sl`, and `slv` to save typing

Sometimes this requires soft-casting/type identifiers with `slv'(foo)`. It seems that the master branch of NVC is unable to determine the difference between `slv` and `std_logic_vector` and `std_ulogic_vector` etc when using concat operators.

Has NVC become more strict with this in v1.14 dev? the NEWS.md makes no mention of anything that stands out?

Thanks

nickg, 2024-11-11:
> Links:

It's possibly a side-effect of 79aef78e319d ("Implement LCS2016_I13 
precedence of unary operators"). Could you try that commit and the one 
before it ec93a1ee5ee9?

a-panella, 2024-11-11:
Interesting results!
`79aef78e319d74f1d8cf744b70c1e6e77190e868` simulated with no issues
`ec93a1ee5ee920cd40d1d249637192aac53d9ba5` simulated with no issues too!

Master still has issues though. Mostly I am after any patches regarding bugfixes for conditional identifiers + external signals that nest into generate statements. My reason for upgrading was hoping for some fixes to fix a random crash I have been observing. I am happy to sit on `79aef78e319d74f1d8cf744b70c1e6e77190e868` if this fixes my crash issues

a-panella, 2024-11-12:
Seems that both these patches work for the ambiguity thing, but my crashes still occur for external signals. Closing this thread as the other crashes are not relevant to operator ambiguity

closed 2024-11-12 by a-panella

nickg, 2024-12-01:
Sorry I forgot to get back to you on this. Are you able to post some examples of the code which is ambiguous now, along with the types of the variables?

a-panella, 2024-12-01:
@nickg I worked around this by just explicitly calling the ambiguous operator. 
```
x <= ieee.std_logic_1164."&"(addr.spi_port_configuration, init_cfg.spi_port_configuration);
```
rather than 
```
x <= addr.spi_port_configuration & init_cfg.spi_port_configuration;
```
where arrays on RHS are subtypes of std_logic_vector like so
```
subtype slv is std_logic_vector;
```
and the type of `addr` and `init_cfg` is like so:
```
type addr_t is record
  spi_port_configuration  : slv(12 downto 0);
  ...
end record;
```

```
type init_cfg_t is record
  spi_port_configuration  : slv(7 downto 0);
  ...
end record;
```

I consider this solved, so I closed it. I don't think you need to look further into this as I've moved on :)
FWIW this concat is performed in a package rather than an entity, I'm not sure if that makes a difference. This particular concatenation triggered the error, but concat elsewhere in my codebase did not. So perhaps package related? I'm not sure, either way I'm happy, the simulator is working nicely again

nickg, 2024-12-03:
It could be related to the use clauses in the package. Do you have `use ieee.std_logic_1164.all`? Did the external name crashes go away on with the latest master branch?

a-panella, 2024-12-03:
@nickg Yes, I tend to always have 
```
use ieee.std_logic_1164.all
```
then typically I have my own package which defines the subtypes of sl and slv internally
```
library work;
use work.synth_pkg.all; --synth_pkg contains `sl` and `slv subtype` declarations
```
Re: external name crashes, my external names were referring to signals that depended on a generate statement. I believe the implementation inside of a generate statement are not processed until elaboration. This meant that my external names were invalid due to generate statement, resulting in a crash (can't remember what type, maybe segfault). It seems you became aware of this and made a fix for this, where the crash no longer happened, and NVC would report the illegal external names and quit instead of crashing. 

This is the fix I got by upgrading my NVC build to master, but the fix that made my sims work again was removing the generate statements entirely, to get around the undefined before elaboration thing. Generates based on a config file were a "nice-to-have" rather than being critical, so I moved on

FWIW I don't think you need to look into this further, ultimately I don't think the simulator is behaving incorrectly, probably the VHDL lang standard has a clause for not using external signals within a generate statement due to when elaboration happens. I'm sure Jim knows if this is something you want more info on, I am no expert on VHDL lang semantics