nickg/nvc #1067: `Fatal: (init): Write to object in frozen arena` and `Caught signal 11 (SEGV_MAPERR)`

Aaron-Hartwig, 2024-11-11:
The code to reproduce this is available on a branch [here](https://github.com/oxidecomputer/quartz/tree/aaron/basic-stream/hdl/ip/vhd/vunit_components/basic_stream). From [this](https://github.com/nickg/nvc/issues/1030#issuecomment-2425332394) it looks like you may have our buck2 environment running already. I'm seeing a couple of different failure modes when attempting to run these simulations. The specifics of the failure, such as which library file is part of the frozen area message, change from run to run. Broadly, the shape of the fails is as such:

```
buck2 run root//hdl/ip/vhd/vunit_components:tb_basic_stream
File changed: root//vunit_out/preprocessed
File changed: root//vunit_out/project_database/0
File changed: root//vunit_out/test_output/test_name_to_path_mapping.txt
20 additional file change events
Build ID: 3795c1d1-6647-44b3-87ed-f73e8b1540bf
Jobs completed: 3. Time elapsed: 0.0s.
BUILD SUCCEEDED
Re-compile not needed

Starting lib.tb_basic_stream.test_single_push_and_pop
Output file: /home/aaron/Oxide/git/quartz/vunit_out/test_output/lib.tb_basic_stream.test_single_push_and_pop_bd13ef0b229f899b341c1941d3b7a96865192fac/output.txt
/usr/bin/nvc --work=lib:/home/aaron/Oxide/git/quartz/vunit_out/nvc/libraries/lib --std=2008 --map=vunit_lib:/home/aaron/Oxide/git/quartz/vunit_out/nvc/libraries/vunit_lib --map=JSON:/home/aaron/Oxide/git/quartz/vunit_out/nvc/libraries/JSON --map=osvvm:/home/aaron/Oxide/git/quartz/vunit_out/nvc/libraries/osvvm --map=lib:/home/aaron/Oxide/git/quartz/vunit_out/nvc/libraries/lib -H 64m -e tb_basic_stream-tb -grunner_cfg=active python runner : true,enabled_test_cases : test_single_push_and_pop,output path : /home/aaron/Oxide/git/quartz/vunit_out/test_output/lib.tb_basic_stream.test_single_push_and_pop_bd13ef0b229f899b341c1941d3b7a96865192fac/,tb path : /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/sims/,use_color : true --no-save --jit -r --dump-arrays --exit-severity=error --ieee-warnings=off
** Warning: design unit LIB.BASIC_STREAM_PKG is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_stream_pkg.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.TH_BASIC_STREAM is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/sims/th_basic_stream.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.BASIC_STREAM_PKG-body is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_stream_pkg.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.BASIC_SOURCE is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_source.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.BASIC_SINK is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_sink.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.TH_BASIC_STREAM-TH is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/sims/th_basic_stream.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.BASIC_SOURCE-MODEL is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_source.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.BASIC_SINK-MODEL is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_sink.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Fatal: (init): Write to object in frozen arena VUNIT_LIB.DATA_TYPES_PRIVATE_PKG-body [address=0x7e5fec000000]
[0x5dbc4c7368a6] 
[0x5dbc4c64e3c0] 
[0x5dbc4c65a464] 
[0x5dbc4c669d28] 
[0x7e5fef44251f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7e5fef5a104a] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x5dbc4c77374f] 
[0x5dbc4c75cf1b] 
[0x5dbc4c75d1fe] 
[0x5dbc4c74e658] 
[0x5dbc4c74e55f] 
[0x5dbc4c7538b0] 
[0x5dbc4c67233e] 
[0x5dbc4c66e7c6] 
[0x5dbc4c671442] 
[0x5dbc4c66e25a] 
[0x5dbc4c666617] 
[0x7e5fef429d8f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7e5fef429e3f] (/usr/lib/x86_64-linux-gnu/libc.so.6) __libc_start_main
[0x5dbc4c667a24] 

nvc 1.14.1 (e10a10e) (Using LLVM 14.0.0) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues

fail (P=0 S=0 F=1 T=4) lib.tb_basic_stream.test_single_push_and_pop (0.2 seconds)
```
```
Starting lib.tb_basic_stream.test_double_push_and_pop
Output file: /home/aaron/Oxide/git/quartz/vunit_out/test_output/lib.tb_basic_stream.test_double_push_and_pop_47aeb18a095b00c3a562b194853d7344e98d4520/output.txt
/usr/bin/nvc --work=lib:/home/aaron/Oxide/git/quartz/vunit_out/nvc/libraries/lib --std=2008 --map=vunit_lib:/home/aaron/Oxide/git/quartz/vunit_out/nvc/libraries/vunit_lib --map=JSON:/home/aaron/Oxide/git/quartz/vunit_out/nvc/libraries/JSON --map=osvvm:/home/aaron/Oxide/git/quartz/vunit_out/nvc/libraries/osvvm --map=lib:/home/aaron/Oxide/git/quartz/vunit_out/nvc/libraries/lib -H 64m -e tb_basic_stream-tb -grunner_cfg=active python runner : true,enabled_test_cases : test_double_push_and_pop,output path : /home/aaron/Oxide/git/quartz/vunit_out/test_output/lib.tb_basic_stream.test_double_push_and_pop_47aeb18a095b00c3a562b194853d7344e98d4520/,tb path : /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/sims/,use_color : true --no-save --jit -r --dump-arrays --exit-severity=error --ieee-warnings=off
** Warning: design unit LIB.BASIC_STREAM_PKG is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_stream_pkg.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.TH_BASIC_STREAM is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/sims/th_basic_stream.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.BASIC_STREAM_PKG-body is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_stream_pkg.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.BASIC_SOURCE is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_source.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.BASIC_SINK is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_sink.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.TH_BASIC_STREAM-TH is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/sims/th_basic_stream.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.BASIC_SOURCE-MODEL is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_source.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check
** Warning: design unit LIB.BASIC_SINK-MODEL is older than its source file /home/aaron/Oxide/git/quartz/hdl/ip/vhd/vunit_components/basic_stream/basic_sink.vhd and should be reanalysed
** Note: you can use the --ignore-time option to skip this check

*** Caught signal 11 (SEGV_MAPERR) [address=0x75e446c00000, ip=0x75e44a3a104a] ***

[0x5ca575e28e9d] 
[0x75e44a24251f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x75e44a3a104a] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x5ca575f3274f] 
[0x5ca575f1bf1b] 
[0x5ca575f1c1fe] 
[0x5ca575f0d658] 
[0x5ca575f0d55f] 
[0x5ca575f128b0] 
[0x5ca575e3133e] 
[0x5ca575e2d7c6] 
[0x5ca575e30442] 
[0x5ca575e2d25a] 
[0x5ca575e25617] 
[0x75e44a229d8f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x75e44a229e3f] (/usr/lib/x86_64-linux-gnu/libc.so.6) __libc_start_main
[0x5ca575e26a24] 

nvc 1.14.1 (e10a10e) (Using LLVM 14.0.0) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues

fail (P=0 S=0 F=2 T=4) lib.tb_basic_stream.test_double_push_and_pop (0.2 seconds)
```
I am happy to help dig in however I can here!

Aaron-Hartwig, 2024-11-11:
It appears to be related to the [`as_stream`](https://github.com/oxidecomputer/quartz/blob/aaron/basic-stream/hdl/ip/vhd/vunit_components/basic_stream/basic_stream_pkg.vhd#L145-L153) function calls in the [testbench](https://github.com/oxidecomputer/quartz/blob/aaron/basic-stream/hdl/ip/vhd/vunit_components/basic_stream/sims/tb_basic_stream.vhd#L36-L37).

Aaron-Hartwig, 2024-11-12:
With a fresh set of eyes today I've found the bug in my test bench that is causing the problem. I'm calling `'range` on a parameter whose type is `natural` [here](https://github.com/oxidecomputer/quartz/blob/aaron/basic-stream/hdl/ip/vhd/vunit_components/basic_stream/sims/tb_basic_stream.vhd#L38). Changing that to be a proper range bound (`data_length(source)-1 downto 0`) then allows everything to run as intended.

Given this is my bug, I'm fine with closing this issue, but I do wonder why the failure mode changed when I was running with debug symbols? Is the abort ignored otherwise? In either case here the failure mode was not very user friendly.

nickg, 2024-11-12:
Now it fails with:

```
** Error: object prefix of attribute RANGE must be an array
    > /home/nick/src/quartz/hdl/ip/vhd/vunit_components/basic_stream/sims/tb_basic_stream.vhd:45
    |
 45 |         variable temp : std_logic_vector(data_length(basic_source_stream)'range);
    |                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |                                          prefix has type NATURAL
```

> Given this is my bug, I'm fine with closing this issue, but I do wonder why the failure mode changed when I was running with debug symbols? 

`--enable-debug` also turns on assertions so it failed earlier than it otherwise would have.

> Is the abort ignored otherwise? In either case here the failure mode was not very user friendly.

Anything that crashes like that with a stack trace is a bug that should be fixed, even if the the input is invalid.

Aaron-Hartwig, 2024-11-12:
Thanks @nickg!

commit 46dc2a88 (2024-11-12): Improve checking of 'RANGE attribute. Fixes #1067