nickg/nvc #1069: Fatal: cannot handle relocation type 6 for symbol ___nvc_do_exit

abaebae, 2024-11-12:
Encountered an issue when running a VUnit test on macOS, which does not occur on Ubuntu (x86_64).

VUnit commit: b0ea8dc4b34c8341bd74ed4de53d4eaca7265360 (vc_standard branch)
Tests run:
``` shell
python vunit/vhdl/verification_components/run.py "*vc_pkg*" -p0 --clean
```

Error log:
```
/opt/homebrew/bin/nvc --work=vunit_lib:/Users/dmw/git/hdl/vunit/vunit_out/nvc/libraries/vunit_lib --std=2008 --map=vunit_lib:/Users/dmw/git/hdl/vunit/vunit_out/nvc/libraries/vunit_lib --map=osvvm:/Users/dmw/git/hdl/vunit/vunit_out/nvc/libraries/osvvm -H 64m -e tb_vc_pkg-a '-grunner_cfg=active python runner : true,enabled_test_cases : Test failing on unexpected message,output path : /Users/dmw/git/hdl/vunit/vunit_out/test_output/vunit_lib.tb_vc_pkg.Test_failing_on_unexpected_message_425b4f93ff7d087781535be29b9cb6f43127adf7/,tb path : /Users/dmw/git/hdl/vunit/vunit/vhdl/verification_components/test/,use_color : true' --no-save --jit -r --exit-severity=error
-- VUNIT_LIB.INTEGER_VECTOR_PTR_PKG.GET(53VUNIT_LIB.INTEGER_VECTOR_PTR_PKG.INTEGER_VECTOR_PTR_TN)I ----
 ff c3 00 d1  fd 7b 02 a9  68 0c 40 b9  49 04 40 f9 |.....{..h.@.I.@.|
 e1 83 00 a9  ff 23 03 29  28 01 80 b9  88 02 f8 b7 |.....#.)(.......|
 09 00 b0 52  1f 01 09 eb  2a 02 00 54  49 00 40 f9 |...R....*..TI.@.|
 4a 08 40 f9  00 32 91 d2  80 09 a0 f2  e1 23 00 91 |J.@..2.......#..|
 29 0d 40 f9  48 a8 00 a9  00 00 cc f2  49 00 00 f9 |).@.H.......I...|
 69 02 80 52  e9 1b 00 b9  08 fc df c8  00 01 3f d6 |i..R..........?.|
 fd 7b 42 a9  ff c3 00 91  c0 03 5f d6  48 7c 00 a9 |.{B......._.H|..|
 08 00 b0 12  09 f6 80 d2  48 7c 01 a9  08 86 80 d2 |........H|......|
 89 6b a0 f2  a8 6b a0 f2  29 00 c0 f2  e1 23 00 91 |.k...k..)....#..|
 28 00 c0 f2  20 01 80 52  48 24 02 a9  c8 01 80 52 |(... ..RH$.....R|
 e8 1b 00 b9  08 00 00 90 >08 01 40 f9  00 01 3f d6 |..........@...?.|
 20 00 20 d4  00 00 00 00  00 00 00 00  00 00 00 00 | . .............|
 b4 00 00 00  00 00 00 03  00 00 00 00  00 00 00 00 |................|
 00 00 00 00  00 00 00 00  10 00 00 00  00 00 00 00 |................|
 01 7a 52 00  01 78 1e 01  10 0c 1f 00  28 00 00 00 |.zR..x......(...|
 18 00 00 00  e4 ff ff ff  ff ff ff ff  b4 00 00 00 |................|
 00 00 00 00  00 44 0e 30  44 9e 01 9d  02 0a 02 60 |.....D.0D......`|
 0e 00 de dd  44 0b 00 00                           |....D...        |
--------------------------------------------------------------------------------
** Fatal: cannot handle relocation type 6 for symbol ___nvc_do_exit
[0x1024382d3] diag_femit+0x9ff
[0x1024aad5f] fatal_trace+0x6f
[0x10247558f] code_load_object+0x2eb
[0x10247cca7] jit_llvm_cgen+0x183
[0x10246039b] jit_tier_up+0x2f
[0x10246c7b3] jit_interp+0x7b
[0x10246d4b7] jit_interp+0xd7f
[0x10246d4b7] jit_interp+0xd7f
[0x10246d4b7] jit_interp+0xd7f
[0x10246d4b7] jit_interp+0xd7f
[0x10246d4b7] jit_interp+0xd7f
[0x10246d4b7] jit_interp+0xd7f
[0x10246d4b7] jit_interp+0xd7f
[0x10245f1e7] jit_try_vcall+0xdb
[0x10245f9eb] jit_try_call+0x77
[0x102435d8f] eval_instance+0x3f
[0x1023da68b] elab_lower+0x7b
[0x1023d9d7b] elab_architecture+0x2d7
[0x1023d91f3] elab+0x55f
[0x10239204b] process_command+0x1f5b
[0x10238fcc7] main+0x41f

nvc 1.14.1 (Using LLVM 19.1.2) [arm-apple-darwin23.6.0]
```

nickg, 2024-11-13:
I can reproduce this but the weird thing is only with the binary from Homebrew, not when I build it myself. It's something to do with how the build system detects the CPU type. I think I've fixed it but I'll need to do a 1.14.2 release before Homebrew picks it up.

abaebae, 2024-11-15:
Thank you for the prompt response! Do you already have a planned release date for version 1.14.2?

nickg, 2024-11-23:
Please try again with the 1.14.2 release which is in homebrew now.

closed 2024-11-23 by nickg

abaebae, 2024-11-23:
Works like a charm. Thanks for the quick fix.