nickg/nvc #1079: write to file failed STD_OUTPUT

mar2345, 2024-11-19:
Hi,
like to share this problem when logging / writing to  STD_OUTPUT
simulation aborts with fatal
first write OK, second write aborts with fatal "write to file failed" 

** Fatal: 0ms+0: write to file failed
   Procedure WRITELINE [TEXT, LINE] at ../lib/std/textio-body.vhd:654
   Procedure LOG [STRING] at test_write_std_out.vhd:15
   Process :test_write_std_out:main at test_write_std_out.vhd:23

test file to reproduce: 

 test_write_std_out.vhd
>>>>>
library std;
use std.textio.all;

entity test_write_std_out is
end entity;

architecture tb of test_write_std_out is

  procedure log(msg : string) is
      variable o : line;
      file OUTPUT: TEXT open WRITE_MODE is "STD_OUTPUT";
      begin
         write(o,string'("::"));
         write(o,msg);
         writeline(output,o);
      end procedure;

begin

  main: process
     begin
        log ("hello world 1");   -- OK 
        log ("hello world 2");   --  abort with fatal 
        log ("hello world 3");
        log ("hello world 4");
        log ("hello world 5");
        wait;

     end process;

end architecture;
<<<<<

test run : 
nvc --version
nvc 1.14-devel (1.13.0.r134.g89e8d452) (Using LLVM 10.0.0)
Copyright (C) 2011-2024  Nick Gasson
...

nvc -a test_write_std_out.vhd
nvc -e test_write_std_out
nvc -r test_write_std_out

::hello world 1
** Fatal: 0ms+0: write to file failed
   Procedure WRITELINE [TEXT, LINE] at ../lib/std/textio-body.vhd:654
   Procedure LOG [STRING] at test_write_std_out.vhd:15
   Process :test_write_std_out:main at test_write_std_out.vhd:23

mar2345, 2024-11-19:
Hi,
just spotted that  
file OUTPUT: TEXT open WRITE_MODE is "STD_OUTPUT";  in procedure log 
defines a new file OUTPUT and HIDES the original  
file OUTPUT : TEXT open WRITE_MODE is "STD_OUTPUT" defined in std package textio 

not really a good idea  ...
if the declaration is removed, the testcase passes with NO problems

procedure log(msg : string) is
variable o : line;
--  remove THIS     file OUTPUT: TEXT open WRITE_MODE is "STD_OUTPUT";  
begin
write(o,string'("::"));
write(o,msg);
writeline(output,o);
end procedure;

so the problem can be solved with the little modification   👍 
 
but still  looks to me there is an unexpected behaviour handling the local file 
shouldn't the file stay open, or be reopened in every call of procedure log ?
writing to it should not fail

Anyhow declaring and using multiple files writing to STD_OUT might not be a great idea, but 
any reason my this should not work ?

nickg, 2024-11-21:
It's a bug with the special case handling of `STD_INPUT` and `STD_OUTPUT`, should be fixed now.

closed 2024-11-21 by nickg

mar2345, 2024-11-22:
Great, a big THANKS for quickly fixing the problem

commit 70ef86f3 (2024-11-21): Do not close STD_INPUT or STD_OUTPUT file handles. Fixes #1079
commit 58eb3dff (2024-11-23): Do not close STD_INPUT or STD_OUTPUT file handles. Fixes #1079