nickg/nvc #1091: [Fuzz] Second iteration

avelure, 2024-12-05:
Following are 13 new crashes from an earlier fuzzing campaign.
For the files with `_standalone` the hash does not match as they had some library dependences and I have minimized them and removed the dependence.

Tested using `nvc 1.15-devel (6ecfc8f) (Using LLVM 15.0.7)` on ubuntu 22.04

[crashes_nvc_new_2.zip](https://github.com/user-attachments/files/18021296/crashes_nvc_new_2.zip)

```
/work/crashes_nvc_new_2/007d677403eec992d943bcd45b40bcac4d961ab3002e10253a47e93374236e59 :
input buffer overflow, can't enlarge buffer because scanner uses REJECT
/work/crashes_nvc_new_2/0c1d01dd6f3aea51f8b086c4c788132486f2bfe6079eb33e89f4e69d39945bc7 :
nvc: ../src/tree.c:998: tree_t tree_ref(tree_t): Assertion `item->object != NULL' failed.

*** Caught signal 6 (SIGABRT) ***

[0x555555613a59] /nvc/build/../src/util.c:900 signal_handler
-->    show_stacktrace();
[0x7ffff0b1351f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7ffff0b679fc] (/usr/lib/x86_64-linux-gnu/libc.so.6) pthread_kill
[0x7ffff0b13475] (/usr/lib/x86_64-linux-gnu/libc.so.6) raise
[0x7ffff0af97f2] (/usr/lib/x86_64-linux-gnu/libc.so.6) abort
[0x7ffff0af971a] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7ffff0b0ae95] (/usr/lib/x86_64-linux-gnu/libc.so.6) __assert_fail
[0x555555727f04] /nvc/build/../src/simp.c:0 simp_tree
[0x55555576c231] /nvc/build/../src/object.c:715 object_rewrite
       for (;;) {
-->       object_t *new = (*ctx->post_fn[object->tag])(object, ctx->context);
          if (new == object || (object = object_rewrite(new, ctx)) == NULL)
[0x55555576bee5] /nvc/build/../src/object.c:779 object_rewrite
                object_t *o = object->items[n].object;
-->             object->items[n].object = object_rewrite(o, ctx);
                object_write_barrier(object, o);
[0x55555576bfdd] /nvc/build/../src/object.c:790 object_rewrite
                      object_t *o = object->items[n].obj_array->items[i];
-->                   if ((o = object_rewrite(o, ctx))) {
                         object_write_barrier(object, o);
[0x5555556b4080] /nvc/build/../src/tree.c:1310 tree_rewrite
-->    object_t *result = object_rewrite(&(t->object), &ctx);
       free(ctx.cache);
[0x555555750ca5] ../src/simp.c:1810 analyse_file
[0x5555555f35b7] analyse
[0x5555555f35b7] /nvc/build/../src/nvc.c:2193 process_command
       case 'a':
-->       return analyse(argc, argv, state);
       case 'e':
[0x5555555f0692] /nvc/build/../src/nvc.c:2355 main
-->    const int ret = process_command(argc, argv, &state);
find: '/usr/local/bin/nvc' terminated by signal 6
/work/crashes_nvc_new_2/4458d32c108c23646b56560b972a18c380feb23e29fd65aab776d729952e77c6_standalone :
nvc: ../src/tree.c:998: tree_t tree_ref(tree_t): Assertion `item->object != NULL' failed.

*** Caught signal 6 (SIGABRT) ***

[0x555555613a59] /nvc/build/../src/util.c:900 signal_handler
-->    show_stacktrace();
[0x7ffff0b1351f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7ffff0b679fc] (/usr/lib/x86_64-linux-gnu/libc.so.6) pthread_kill
[0x7ffff0b13475] (/usr/lib/x86_64-linux-gnu/libc.so.6) raise
[0x7ffff0af97f2] (/usr/lib/x86_64-linux-gnu/libc.so.6) abort
[0x7ffff0af971a] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7ffff0b0ae95] (/usr/lib/x86_64-linux-gnu/libc.so.6) __assert_fail
[0x555555727f04] /nvc/build/../src/simp.c:0 simp_tree
[0x55555576c231] /nvc/build/../src/object.c:715 object_rewrite
       for (;;) {
-->       object_t *new = (*ctx->post_fn[object->tag])(object, ctx->context);
          if (new == object || (object = object_rewrite(new, ctx)) == NULL)
[0x55555576bee5] /nvc/build/../src/object.c:779 object_rewrite
                object_t *o = object->items[n].object;
-->             object->items[n].object = object_rewrite(o, ctx);
                object_write_barrier(object, o);
[... 300 lines cut ...]
          if (new == object || (object = object_rewrite(new, ctx)) == NULL)
[0x55555576bee5] /nvc/build/../src/object.c:779 object_rewrite
                object_t *o = object->items[n].object;
-->             object->items[n].object = object_rewrite(o, ctx);
                object_write_barrier(object, o);
[0x55555576bfdd] /nvc/build/../src/object.c:790 object_rewrite
                      object_t *o = object->items[n].obj_array->items[i];
-->                   if ((o = object_rewrite(o, ctx))) {
                         object_write_barrier(object, o);
[0x5555556b4080] /nvc/build/../src/tree.c:1310 tree_rewrite
-->    object_t *result = object_rewrite(&(t->object), &ctx);
       free(ctx.cache);
[0x555555750ca5] ../src/simp.c:1810 analyse_file
[0x5555555f35b7] analyse
[0x5555555f35b7] /nvc/build/../src/nvc.c:2193 process_command
       case 'a':
-->       return analyse(argc, argv, state);
       case 'e':
[0x5555555f0692] /nvc/build/../src/nvc.c:2355 main
-->    const int ret = process_command(argc, argv, &state);
/work/crashes_nvc_new_2/f0cfccae8feeecc7b26d25c107709a99fb0f1d62f901b0430813e84c30f78382 :
fatal: tree kind T_PSL does not have item I_FLAGS
/work/crashes_nvc_new_2/fc2d4e66b3ea007a59d8becdf1e6eb8a8881ff2eb3f90cd5667bf3f19fcc3471 :
find: '/usr/local/bin/nvc' terminated by signal 11
```

NikLeberg, 2024-12-09:
Some tracking information to have an overview of crashes and fixes:

| Crash File Hash | State | PR |
| --- | --- | --- |
| `0c1d01dd6f3aea51f8b086c4c788132486f2bfe6079eb33e89f4e69d39945bc7` | ✔️ done | #1092 |
| `007d677403eec992d943bcd45b40bcac4d961ab3002e10253a47e93374236e59` | error `input buffer overflow` in ylex, won't fix? | - |
| `8bfa1e5ce26fbc6d552138c413ce92756ca99bc61b002b567256668c4aaf1315` | ⏳ analyzing | - |
| `51c4a1798e1f1aeff3711fc660cc9fa9a06c8ab38777c4fb40e79b23e5682add_standalone` | ✔️ done | #1093 |
| `87be50d057c5d498033f6e63a432d3a1aefbbfd77f8fcfd45c91b54334636c10` |  duplicate of `8bfa1e5` | n/a |
| `92f257919da203705a3648c0617fd2ccb179689e7ba64d033e7a556e448da488` | duplicate of `8bfa1e5` | n/a |
| `4458d32c108c23646b56560b972a18c380feb23e29fd65aab776d729952e77c6_standalone` | duplicate of `51c4a17` | n/a |
| `9618d8b1821db86af1bbce73263f830ebd77f2525a7ed3b1d20439d8fee97e16` | duplicate of `51c4a17` | n/a |
| `a04900efaed84ebf347d7c8325e5bf9fd96b67deb31508d4149b17b558e703c0_standalone` | - | - |
| `b8c6cfce22dd64da3b67f277b7ec3bff6ed67e368dc47c3f7fa91a2b75751fc3_standalone` | - | - |
| `da346642705230c6f2c89b7c0a522d1078c44b3d0dfd5fe52e2859212edd2784` | - | - |
| `f0cfccae8feeecc7b26d25c107709a99fb0f1d62f901b0430813e84c30f78382` | - | - |
| `fc2d4e66b3ea007a59d8becdf1e6eb8a8881ff2eb3f90cd5667bf3f19fcc3471` | - | - |

NikLeberg, 2025-01-12:
@nickg Ignoring the _won't fix_ crashes, the remaining crashes are a number too big for me. They all revolve around PSL and I think are connected to #1001.

nickg, 2025-01-19:
> [@nickg](https://github.com/nickg) Ignoring the _won't fix_ crashes, the remaining crashes are a number too big for me. They all revolve around PSL and I think are connected to [#1001](https://github.com/nickg/nvc/issues/1001).

Thanks for going through these. The crashes in #1001 should be fixed now (or at least just produce errors).

NikLeberg, 2025-01-24:
Can confirm that the PSL crashes no longer happen. Those PSL changes look like they took a lot of work and thought. I love how elegantly you did them.

Alright, so I think this could be closed? The remaining two errors are the buffer overflow in the lexer (which already gives a reasonable error output) and a stack overflow (because `(` is spammed in an expression). If you'd like the stack overflow to be fixed I could think about implementing some recursion level _cutoff_ for expressions or something like that.

jmgc, 2025-08-28:
I have tried to compile with nvc 1.17.2 (using LLVM 15.0.7) a VHDL code that works on ghdl and vcom. I also get the cryptic message:

"input buffer overflow, can't enlarge buffer because scanner uses REJECT
Analysis failed"

I understand it comes from the lexer.

Is there any way to have more information about what is happening? At least the line in the code that is causing it?

Thanks in advance.

nickg, 2025-08-28:
> Is there any way to have more information about what is happening? At least the line in the code that is causing it?

It happens when there is a single input token longer than Flex's default buffer size (16kB).

The latest master branch should print the location where the error occurs.