nickg/nvc #1097: vhpi_register_cb does shallow copy of callback data

svnesbo, 2024-12-11:
It seems like only a shallow copy of callback data is done in `vhpi_register_cb` in `src/vhpi/vhpi-model.c`:

```
cb->data    = *cb_data_p;
```

But in the VHDL 2008 LRM, under 21.2.2 Registering callbacks, it says (my emphasis):
> After the vhpi_register_cb function returns, the tool does not retain any references to the callback data
structure _or to storage pointed to by members of the callback data structure._

My naive interpretation of this is that it's necessary to do a deep copy of the `time`, `value`, and `user_data` fields in the callback data struct. But I'm kind of new to VHPI so maybe I don't understand how this datatype is supposed to be used.
```
typedef struct vhpiCbDataS
{
  int32_t reason;            /* callback reason */
  void (*cb_rtn) (const struct vhpiCbDataS *);  /* call routine */
  vhpiHandleT obj;             /* trigger object */
  vhpiTimeT *time;             /* callback time */
  vhpiValueT *value;           /* trigger object value */
  void *user_data;             /* pointer to user data to be passed
                                  to the callback function */
} vhpiCbDataT;
```

nickg, 2024-12-12:
We can't make a deep copy of `user_data` as there's no way for the VHDL implementation to know how large the object is. The `time` field I _think_ is OK at the moment since we always replace it with a pointer to a `vhpiTimeT` on the stack before calling the callback (i.e. the user's `time` value is ignored once `vhpi_register_cb` returns). The `value` field is not used at the moment. Have you got a case where the shallow copy causes a problem?

svnesbo, 2024-12-12:
I don't have a real case at the moment, this was just something that caught my eye. I'm learning/experimenting with VHPI and chose to use NVC for that. I know NVC doesn't promise full VHPI support, but enough to run Cocotb I read somewhere, which is probably sufficient for me.

You're right about `user_data`, pretty much have to pass along the pointer for that one. But what initially confused me was actually the `time` field. Based on [this example](https://github.com/ghdl/ghdl/blob/master/testsuite/vhpi/005trace_file/vhpi_lib.c) in the GHDL repo I was trying to make a callback for when the simulation starts and ends. I added an example below.

```
#include <vhpi_user.h>

void start_of_sim_cb(const vhpiCbDataT * cb_data) {
  vhpi_printf("Start of simulation\n");
  if (cb_data->time != NULL) {
    vhpi_printf("Time = %d\n", cb_data->time->low);
  }
}

void end_of_sim_cb(const vhpiCbDataT * cb_data) {
  vhpi_printf("End of simulation.\n");
  if (cb_data->time != NULL) {
    vhpi_printf("Time = %d\n", cb_data->time->low);
  }
}

void startup_1()
{
  vhpiCbDataT cb_data;

  static vhpiTimeT t = {.high = 0, .low = 0};

  vhpi_printf("startup_1 running\n");

  cb_data.obj = NULL;
  cb_data.time = &t;
  cb_data.value = NULL;
  cb_data.user_data = NULL;

  cb_data.reason = vhpiCbStartOfSimulation;
  cb_data.cb_rtn = start_of_sim_cb;
  vhpi_register_cb(&cb_data, 0);

  t.low = 1234;
  cb_data.reason = vhpiCbEndOfSimulation;
  cb_data.cb_rtn = end_of_sim_cb;
  vhpi_register_cb(&cb_data, 0);
}

void (*vhpi_startup_routines[])() = {
  startup_1,
  NULL
};
```

```
entity tb is
end entity tb;

architecture sim of tb is
begin
  p_test : process
  begin
    report "Starting simulation";
    wait for 200 ns;
    report "Simulation done";
    std.env.stop;
  end process;
end architecture sim;
```

I naively thought I could get the timestamp for when the callbacks were triggered from `cb_data->time` passed to the callback routines. Of course it doesn't make much sense to set a value for the time when the callback is registered (it was like that in the original example), but I figured the time value would be replaced with the actual time. And obviously the start of sim time should be zero, but the end of sim time is interesting to know.

If I run that code above then this is the output I get:
```
** Note: loading VHPI plugin ./vhpi_test_lib.so
** Note: startup_1 running
** Note: (init): Start of simulation
** Note: (init): Time = 1234
** Note: 0ms+0: Starting simulation
   Process :tb:p_test at vhpi_test.vhd:11
** Note: 200ns+0: Simulation done
   Process :tb:p_test at vhpi_test.vhd:11
** Note: 200ns+0: STOP called
   Procedure STOP [] at ../lib/std.08/env-body.vhd:32
   Process :tb:p_test at vhpi_test.vhd:19
** Note: 200ns+0: End of simulation.
** Note: 200ns+0: Time = 1234
```

So, initially I was puzzled that it just printed the value I supplied when I registered the callback. I was also confused by the fact that in the original example (and others I've seen) they referenced a local `vhpiTimeT` variable when they registered the callback (I made it static above).

But I've read the LRM more carefully now and I realize it says that only for "object callbacks" (callbacks related to signal and variable events) should it replaced the time value in the callback data with the time of the event. And only if time is not NULL when the callback is registered (it doesn't matter what it points to and the pointer doesn't have to be valid, it will be replaced). I see now that this is implemented for signal-callbacks in `vhpi_signal_event_cb` in NVC, and not for non-signal callbacks in `vhpi_global_cb` (and it's not supposed to). If I want the timestamp in a non-signal callback I discovered I can just use `vhpi_get_time` instead.

So, I figured out what I needed. But I feel like maybe it's an issue that NVC blindly passes the `cb_data_p->time` pointer to those callbacks. If I change `t` to be a non-static local variable that goes out of scope after the startup routine, then that `cb_data->time` pointer is invalid and some random junk will be printed. Personally, I would have preferred if `vhpi_register_cb` just set `time` to NULL for any callback reason that doesn't use that field or maybe even issue a warning/error if it was not NULL in `cb_data_p` when it was called.
But I can't make sense of what's the correct behavior according to the (2008) LRM, it seems to contradict itself on this point. Under 21.3.1 it says *If a VHPI program provides a pointer to a time structure in the time member of a callback data structure, the VHPI program shall allocate the memory for the time structure*. Ok, so "VHPI program" I assume is the user code, so that means it's the user's responsibility to allocate memory for `time`. Does it also imply that the user is free to use this field to pass data to the callback when it has no other purpose? But that's not how it's used for object events, then it explicitly says the tool shall allocate mem for `time`. While for more general callbacks not tied to signals/events, under 21.3.6, 21.3.7, and 21.3.8, it says *The values of the obj, time, and value members of the callback data structure for an action callback are ignored by the tool.*. 🤷

Sorry about the long post 😄 I don't know if this is an issue, you can probably close it.

nickg, 2024-12-14:
> Personally, I would have preferred if vhpi_register_cb just set time to NULL for any callback reason that doesn't use that field or maybe even issue a warning/error if it was not NULL in cb_data_p when it was called.

This seems reasonable to me. I agree the LRM isn't very clear on what should happen here.

commit 639af431 (2024-12-14): Warn if VHPI non-object callback has non-NULL obj/value/time fields

Fixes #1097
closed 2024-12-14 by nickg