nickg/nvc #1098: Speed-up compilation of Vivado libraries

tasgomes, 2024-12-11:
I use the following command to compile the Vivado libraries:

` nvc --install vivado xpm_vhdl --dest=<some_directory>`

This command also compiles the `xpm_vhdl` components but this is out of scope for this problem.

I noticed that NVC compiles each component individually, i.e., NVC is called each time a new file is compiled. I think this creates a big overhead during the compilation of the libraries.

With Questa or Modelsim, we can do the same but performance (the time it takes to compile everything) is severely affected.

A much better approach is to list all the components in a file and provide this file to a single call of the `vcom` command, e.g.:

`vcom (...) -f <file_with_all_lib_components_listed>.cmf`

This approach is really really much faster than calling `vcom` for each file.

I do use a Questa version that is not supported by Vivado so I have to compile the libraries manually (i.e., not using Vivado) and I can see the big different with both approaches.

To give some context, I do enjoy the speed of NVC during simulation compared to Questa. In our runners setup, we always compile the Vivado libraries before running simulation. As this takes less than 1 minute with Questa, the time cost is very minimal so we do it all the time. But with NVC, this is taking almost 8 minutes to finish. 

At the end, running the runner job with questa ends up being faster than NVC just because NVC takes a lot of time to compile the libraries, which is a bummer.

Does NVC support such a feature where we can compile all files in one go?

nickg, 2024-12-11:
> Does NVC support such a feature where we can compile all files in one go?

Yes there is the `-f` option when analysing that does exactly this. I changed the Vivado install script to use this and it's around 2x faster on my machine. If you only want the 2008 libraries you can do `nvc --std=2008 --install vivado` which will be another 2x faster again.

closed 2024-12-11 by nickg

nickg, 2024-12-13:
I looked at why this is so slow and found an embarrassing bug. With the latest master `nvc --install vivado` now takes around four seconds on my machine.

tasgomes, 2024-12-13:
> latest master branch nvc --install vivado now takes around four seconds

Wow that is really good and 4 seconds is outstanding! Thanks for improving
this.

commit 72b5e72f (2024-12-11): Pass Vivado vhdl_analyze_order to nvc -a -f directly. Fixes #1098