nickg/nvc #1112: Impure function in package generics

amb5l, 2024-12-23:
I'm seeing the following error compiling OSVVM 2024.07 source:
```
** Error: unexpected impure while parsing interface declaration, expecting one of constant, signal, variable, file, identifier, type, function, procedure or package
    > D:\work\tyto2\submodules\OsvvmLibraries\osvvm\ScoreboardGenericPkg.vhd:99
    |
 99 |     impure function Match(Actual : ActualType ;                           -- defaults
    |     ^^^^^^ this token was unexpected
```
Perhaps NVC does not support impure functions as package generics?

I built NVC from today's master branch:
`nvc 1.13-devel (1.12.0.r4.ge7190884) (Using LLVM 17.0.6)`

nickg, 2024-12-23:
> 
> I built NVC from today's master branch: `nvc 1.13-devel (1.12.0.r4.ge7190884) (Using LLVM 17.0.6)`

Are you sure that's the latest master branch? e7190884 is from April this year. I think this was fixed in #917.

amb5l, 2024-12-23:
You're right, apologies for wasting your time. My build process was broken. Should have noticed the old tag in the version string!

closed 2024-12-23 by amb5l