nickg/nvc #1113: Library format / version issue

amb5l, 2024-12-23:
When compiling OSVVM 2024.07 sources, NVC emits a library format error, as captured below:

```
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/IfElsePkg.vhd
** Warning: directory osvvm already exists and is not an NVC library
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/OsvvmScriptSettingsPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/OsvvmSettingsPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/TextUtilPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/ResolutionPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/NamePkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/OsvvmGlobalPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/VendorCovApiPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/LanguageSupport2019Pkg_c.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/TranscriptPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/AlertLogPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/TbUtilPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/NameStorePkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/MessageListPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/SortListPkg_int.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/RandomBasePkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/RandomPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/RandomProcedurePkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/CoveragePkg.vhd
** Warning: declaration of I hides an earlier declaration with the same type
      > D:\work\tyto2\submodules\OsvvmLibraries\osvvm\CoveragePkg.vhd:2736
      |
 2734 |       for i in 1 to 20 loop
      |           ^ this declaration is hidden
 2735 |         if CovStructPtr(ID.ID).FieldName /= NULL then
 2736 |           for i in 1 to CovStructPtr(ID.ID).FieldName'length loop
      |               ^ by the constant declaration here
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/DelayCoveragePkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/ClockResetPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/ResizePkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/ScoreboardGenericPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/ScoreboardPkg_slv.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/ScoreboardPkg_int.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/ScoreboardPkg_signed.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/ScoreboardPkg_unsigned.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/ScoreboardPkg_IntV.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/MemorySupportPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/MemoryGenericPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/MemoryPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/ReportPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/OsvvmTypesPkg.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/OsvvmScriptSettingsPkg_default.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/OsvvmSettingsPkg_default.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm:osvvm -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/osvvm/OsvvmContext.vhd
cd sim_nvc && nvc -H 128M --std=2008 --work=osvvm_common:osvvm_common -a --relaxed D:/work/tyto2/submodules/OsvvmLibraries/Common/src/ModelParametersPtPkg.vhd
** Warning: directory osvvm_common already exists and is not an NVC library
** Fatal: D:\work\.nvc\lib\OSVVM\OSVVM.OSVVMCONTEXT: serialised format digest is 8042b8f0 expected 4d1a7bb3. This design unit uses a library format from an earlier version of nvc and should be reanalysed.
```

Note that this occurs as soon as I begin compiling into a different library (osvvm_common).

My NVC binary is built (really!) from today's master: `nvc 1.15-devel (1.14.0.r206.g5cca94bb) (Using LLVM 17.0.6)`

nickg, 2024-12-23:
> `nvc -H 128M --std=2008 --work=osvvm_common:osvvm_common -a  ...`

Try passing `-L.` here. I think you have another copy of OSVVM built with an older version in `D:\work\.nvc\lib\osvvm` which is on the search path by default whereas the one you just built in `.\osvvm` isn't.

amb5l, 2024-12-24:
Thanks Nick and apologies for 2 dumb mistakes in 1 day! NVC runs my simulation drastically faster than GHDL and ModelSim, which is what I was after.

closed 2024-12-24 by amb5l

nickg, 2024-12-28:
> NVC runs my simulation drastically faster than GHDL and ModelSim, which is what I was after.

Can you test again with the latest master branch? I just pushed some more performance improvements today.

amb5l, 2024-12-29:
The simulation - as it now stands - runs so quickly (a few seconds) that makefile processing and other OS stuff wrapped around it is giving me a poor SNR when I run a benchmark.  That said, NVC is >20% faster than GHDL at runtime and maybe twice as fast or better for compilation and elaboration. The new master does not *seem* to improve this. I have some other work coming up that may be more useful for performance analysis, I will let you know when that is ready to look at.