nickg/nvc #1134: wait until in for loop waits only once

m-kru, 2025-01-15:
`nvc 1.16-devel (1.15.0.r4.g230dffd9) (Using LLVM 14.0.0)`

I am not 100 % sure it is a bug, but it smells like a bug.
https://github.com/m-kru/vhdl-amba5/blob/c140cce127754a53599bdf802ed55bac7a5e5eab/apb/tb/serial-bridge/tb-read.vhd#L106

I would expect this loop to wait for 4 rising edges with valid handshake conditions.
What is observe is that it iterates 4 items after the
`wait until rising_edge(clk) and byte_out_ready = '1' and sb.byte_out_valid = '1';`
becomes true for the first time.

If I manually unroll the loop to:
```vhdl
      -- Read data
      byte_out_ready <= '1';
      wait until rising_edge(clk) and byte_out_ready = '1' and sb.byte_out_valid = '1';
      rdata(31 downto 24) <= sb.byte_out;
      byte_out_ready <= '0';
      wait for delay * CLK_PERIOD;

      byte_out_ready <= '1';
      wait until rising_edge(clk) and byte_out_ready = '1' and sb.byte_out_valid = '1';
      rdata(23 downto 16) <= sb.byte_out;
      byte_out_ready <= '0';
      wait for delay * CLK_PERIOD;

      byte_out_ready <= '1';
      wait until rising_edge(clk) and byte_out_ready = '1' and sb.byte_out_valid = '1';
      rdata(15 downto 8) <= sb.byte_out;
      byte_out_ready <= '0';
      wait for delay * CLK_PERIOD;

      byte_out_ready <= '1';
      wait until rising_edge(clk) and byte_out_ready = '1' and sb.byte_out_valid = '1';
      rdata(7 downto 0) <= sb.byte_out;
      byte_out_ready <= '0';
      wait for delay * CLK_PERIOD;

      assert rdata = want
        report "invalid rdata, got " & rdata'image & ", want " & want'image
        severity failure;

```
it works as expected. I catch 4 bytes on rising edges with valid handshake condition.

If you clone the repo you can run simulation by executing the following commands:
```
nvc  --std=2019 -L. --work=apb  -a ./apb.vhd
nvc  --std=2019 -L. --work=apb  -a ./checker.vhd
nvc  --std=2019 -L. --work=apb  -a ./mock-completer.vhd
nvc  --std=2019 -L. --work=apb  -a ./serial-bridge.vhd
nvc  --std=2019 -L. --work=work -a ./tb/serial-bridge/tb-read.vhd
nvc  --std=2019 -L. -e tb_read
nvc --messages=compact --std=2019 -L. -r tb_read --wave --exit-severity=error --dump-arrays
```

nickg, 2025-01-15:
```
for i in 3 to 0 loop
```

This executes zero times (`3 to 0` is a null range, you probably want `3 downto 0`).

amb5l, 2025-01-15:
I use null ranges in a few situations. I do not like the way they are specified in VHDL, and I can well understand your frustration, but I would not want existing working code to start causing compilation warnings.

nickg, 2025-01-15:
> I use null ranges in a few situations. I do not like the way they are specified in VHDL, and I can well understand your frustration, but I would not want existing working code to start causing compilation warnings.

How about in for loops specifically? Modelsim does issue a warning in this case:

```
** Warning: ../test/bounds/issue1134.vhd(9): (vcom-1246) Range 3 to 0 is null.
** Warning: ../test/bounds/issue1134.vhd(11): (vcom-1246) Range 3 to 0 is null.
** Warning: ../test/bounds/issue1134.vhd(16): (vcom-1246) Range 3 to 0 is null.
```

amb5l, 2025-01-15:
> How about in for loops specifically?

Good point - no. If ModelSim issues a warning for this case then I'd support NVC doing so.

avelure, 2025-01-15:
Modelsim warns on all null ranges with the same 1246 warning, which I find quite annoying, but I see that it would have helped in this situation.

m-kru, 2025-01-16:
> I use null ranges in a few situations. I do not like the way they are specified in VHDL, and I can well understand your frustration, but I would not want existing working code to start causing compilation warnings.

When do you use null ranges? I have never found any use case for them.

amb5l, 2025-01-16:
> When do you use null ranges? I have never found any use case for them.

I've tried and failed to locate the relevant code, but - from memory.. there are times when binding an unconstrained vector port in a module to a null range signal in its parent indicates that port is not needed (I don't think a module can tell if an output is left `open`), and it then adjusts itself accordingly (via `if generate` constructs). Sorry if this is not helpful!

m-kru, 2025-01-17:
Do you have any open source example?

amb5l, 2025-01-18:
See [this file](https://github.com/amb5l/tyto2/blob/main/src/common/basic/intel/ram_tdp_ar.vhd) for an example.

In this case I have a generic - a 2D array of std_logic - that can be used to initialise the contents of inferred RAM blocks. Its default value is a null range. A function is used to detect the dimensions of the generic and skip initialisation if it doesn't match the RAM dimensions. This is not a great example, but was the best I could find.

tasgomes, 2025-03-14:
@m-kru There is an example of null ranges in OSVVM. In fact, I pushed a fix, some days ago, in there to remove the "unnecessary" warning on Modelsim/Questa:

https://github.com/OSVVM/OSVVM/pull/102/commits/a474878618a0fcf6c18ed210f3257cf2cfb695be

oscargus, 2025-04-25:
Related request for vhdl-ls (so more people want to have warning for this): https://github.com/VHDL-LS/rust_hdl/issues/364