nickg/nvc #1135: Embedded PSL parsing

amb5l, 2025-01-15:
When compiling [this source file](https://github.com/user-attachments/files/18429966/memac_rmii_rx_spd.zip) as follows...
```
nvc -a -psl memac_rmii_rx_spd.vhd
```
...NVC emits the following error message:
```
** Error: unexpected identifier while parsing PSL Braced SERE, expecting one of ** or }
     > D:\work\tyto2\src\common\ethernet\memac_rmii_rx_spd.vhd:394
     |
 394 | end architecture rtl;
     |     ^^^^^^^^^^^^ this token was unexpected
```
See line 387:
```
  -- psl assert always (
```
If this is replaced with these 2 lines...
```
  -- psl
  --  assert always (
```
...then the error goes away.

I am using NVC on Windows/MSYS:
```
nvc 1.16-devel (1.15.0.r4.g230dffd9) (Using LLVM 19.1.6)
```

nickg, 2025-01-18:
I've fixed the problem with newlines but there's still an issue with parsing `prev(..)` inside VHDL expressions. I think this is technically not allowed by the PSL grammar but seems to be widely supported.

Blebowski, 2025-01-19:
@nickg this is little bit trickily said in the PSL 2010 spec, sec. 5.2:

```
In each flavor of PSL, at any place where an HDL subexpression may appear within an HDL or PSL
expression, the grammar of the corresponding HDL is extended to allow any form of HDL or PSL
expression. Thus HDL expressions, PSL expressions, built-in functions, and union expressions may all be
used as subexpressions within HDL or PSL expressions.
```

E.g. `prev` is specified to have arbitrary type:
```
NOTE 2—The first argument of prev() is not necessarily a Boolean expression. For example, if the argument to
prev() is a bit vector, then the result is the previous value of the entire bit vector.
```
if `Built_In_Function_Call` was only allowed to be as standalone `HDL_or_Psl_Expression`, then there
would be no place where such call could be used, since top-most expressions in sequence / SERE or
clock context need to be Boolean or Bit.

commit 2c4d8c5c (2025-01-19): Implement PSL prev() built-in function. Fixes #1135
closed 2025-01-19 by nickg