nickg/nvc #1143: Array indexed by enumerated type results in '-' or 'u' signal

McSherry, 2025-01-22:
I'm using nvc 1.14.2.

I have a signal that is an array of arrays, where the first level is indexed by an enumerated type:

```vhdl
type reg_e is (
    I2CCFG,
    I2CCON,
    I2CDR,
    -- ...and so on
);
    
type   read_port_t is array(reg_e range reg_e'high downto reg_e'low) of std_ulogic_vector(apb_prdata_o'range);
signal read_port     : read_port_t;
```

This signal is connected to the output of the block (which is an APB completer) through a pipelined mux tree block:

```vhdl
prdata_mux: entity work.mux
generic map(
    Ports          => reg_e'pos(reg_e'right) + 1,
    Width          => apb_prdata_o'length,
    InputRegister  => false,
    SelectRegister => false,
    OutputRegister => true,
    Stages         => 2
)
port map(
    clk_i => clk_i,
    sel_i => address_as_select,
    in_i  => read_port_to_slm2(read_port),
    out_o => apb_prdata_o
);
```

Here, `address_as_select` is a function that boils down to a `case` statement that translates the APB PADDR signal into a `reg_e` value, and the function `read_port_to_slm2` is a trivial copy to a similarly-shaped type we use common across our IP cores:

```vhdl
type std_ulogic_matrix2 is array(natural range <>) of std_ulogic_vector;

impure function read_port_to_slm2(rp : read_port_t) return std_ulogic_matrix2 is
    variable rv : std_ulogic_matrix2(reg_e'pos(reg_e'right) downto 0)(apb_prdata_o'range);
begin
    for reg in reg_e'left to reg_e'right loop
        rv(reg_e'pos(reg)) := rp(reg);
    end loop;

    return rv;
end function read_port_to_slm2;
```

Arranged like this, in simulation with nvc, the output on `apb_prdata_o` is constant at a partially don't-care, partially undefined value. The way the block is defined, whether or not an APB read is ongoing, the `apb_prdata_o` signal has the value of the currently-addressed register on it, so it should not remain constant. Here is a waveform showing this behaviour (five APB writes followed by an APB read):

[image]

I initially thought this might be an issue with the pipelined mux tree. The first two APB writes set up a strobe generator, and the two `strobes_g` rows near the bottom of the image come from that strobe generator through another instantiation of the same mux tree component. As these work fine, it suggests that the mux tree is not the issue.

I thought from there it might be an issue with using an enumerated type as an index, so I changed the definition of `read_port` so that it used a `to` range instead of a `downto` to see if it made a difference:

```vhdl
type read_port_t is array(reg_e range reg_e'left to reg_e'right) of std_ulogic_vector(apb_prdata_o'range);
```

(I also changed `read_port_to_slm2` to use `to` ordering as well)

While this didn't solve the issue, it did change the observed behaviour&mdash;the constant-but-wrong output on `apb_prdata_o` is now different:

[image]

To me, this suggests it's something with how nvc is handling the `read_port_t` type.

nickg, 2025-01-25:
It's a bit difficult for me to work out what's going on here from the description. Are you able to reduce it to a smaller self-contained example?

> sel_i => address_as_select,

Is `address_as_select` a function that takes no arguments?

NikLeberg, 2025-05-21:
I guess I found a reproducer for an at least related bug. If the entity instance port is connected to a function, that function seems to only be evaluated once. At least in this case.

```vhd
library ieee;
use ieee.std_logic_1164.all;

package pkg is
    type reg_e is (
        I2CCFG,
        I2CCON,
        I2CDR
    );
end package;

--------------------------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use work.pkg.all;

entity mux is
    port (
        reg_in  : in reg_e;
        reg_out : out reg_e
    );
end entity;

architecture arch of mux is
begin
    reg_out <= reg_in;
end architecture;

--------------------------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use work.pkg.all;

entity issue1143 is
end entity;

architecture bug of issue1143 is
    signal sel : natural;
    signal reg_out, reg_direct : reg_e;
begin

    mux_inst : entity work.mux
        port map (
            reg_in => reg_e'val(sel), -- this seems to be evaluated only once
            reg_out => reg_out
        );

    reg_direct <= reg_e'val(sel); -- this is evaluated/updated as expected

    test : process is
    begin
        sel <= 0;
        wait for 1 ns;
        report "out: " & to_string(reg_out) & ", direct: " & to_string(reg_direct);

        sel <= 1;
        wait for 1 ns;
        report "out: " & to_string(reg_out) & ", direct: " & to_string(reg_direct);

        sel <= 2;
        wait for 1 ns;
        report "out: " & to_string(reg_out) & ", direct: " & to_string(reg_direct);

        wait;
    end process;

end architecture;
```

Output:
```
** Note: 1ns+0: out: i2ccfg, direct: i2ccfg
   Process :issue1143:test at ../test/elab/issue1143.vhd:56
** Note: 2ns+0: out: i2ccfg, direct: i2ccon
   Process :issue1143:test at ../test/elab/issue1143.vhd:60
** Note: 3ns+0: out: i2ccfg, direct: i2cdr
   Process :issue1143:test at ../test/elab/issue1143.vhd:64
```

Issue being, that the "out" value is not changing.

NikLeberg, 2025-05-21:
Note: Same bug if the port actual is a function with no arguments. If the function takes arguments it works fine.

```vhd
architecture bug of issue1143 is

    -- [...]

    function f_selector(i : natural) return reg_e is begin
        return reg_e'val(i);
    end function;

    impure function f_selector_no_args return reg_e is begin
        return reg_e'val(sel);
    end function;
begin

    mux_inst : entity work.mux
        port map (
            -- reg_in => reg_e'val(sel),     -- bug, value not changing
            -- reg_in => f_selector_no_args, -- bug, value not changing
            reg_in => f_selector(sel),       -- working fine
            reg_out => reg_out
        );

    -- [...]

end architecture;
```

nickg, 2025-05-21:
I've fixed the case where the signal is a parameter to an attribute reference. The case where it's read inside an impure function call also seems to be valid according to the LRM, although only if the function body is in the same design unit as the `process (all)`. But none of the simulators I tried support it:

```vhdl
entity issue1143 is
end entity;

architecture test of issue1143 is
    signal x : character;

    impure function get_pos return integer is
    begin
        return character'pos(x);
    end function;
begin

    b: block is
        port (
            p1, p2 : in integer );
        port map (
            p1 => character'pos(x),
            p2 => get_pos );
    begin

        check: process is
        begin
            assert p1 = integer'left;
            assert p2 = integer'left;
            wait for 0 ns;
            assert p1 = 0;
            assert p2 = 0;
            wait for 0 ns;
            assert p1 = 97;
            assert p2 = 97;
            wait for 1 ns;
            wait for 0 ns;
            assert p1 = 98;
            assert p2 = 98;
            wait;
        end process;

    end block;

    x <= 'a', 'b' after 1 ns, 'c' after 2 ns;

end architecture;
```

The `p2` assertions fail on GHDL, Aldec, and Questa.

NikLeberg, 2025-05-22:
Although I'm not a LRM expert, I agree with your argumentation. Would there be a benefit of generating a warning message?

Would be interesting to know if @McSherry's issue is fixed with this.

closed 2025-07-14 by nickg