nickg/nvc #1145: When accessing out of range on a custom type ,SIM Crashes  with Caught signal 11 (SEGV_MAPERR)

alpkoyun, 2025-01-23:
Hi,

I have encountered a bug where simulation craches with " Caught signal 11 (SEGV_MAPERR)". It gives no other error message syntax issue etc. I tracked down the cause to accessing out of range on a custom type like below.  In the code below, the range of state_array is 0 to "read_size -1" . In the for  generate, the range 0 to "write_size - 1" is tried to be accessed. when "read_size" is smaller than the "write_size" simulation chashes. I tried the same for the variable "test_out" but it gave the correct range warnings and did not crash. I noticed that the simulation does not crash if you remove the "reset" on the process.

`
library IEEE;
use IEEE.STD_LOGIC_1164.all;
use IEEE.NUMERIC_STD.all;
library work;

entity nvc_bug is
  generic (
    read_size    : integer := 2;
    write_size    : integer := 16
  );

  port (
    clk : in std_logic;
    reset : in std_logic;

    write_en     : in std_logic_vector(write_size -1 downto 0);
    test_out : out std_logic_vector(write_size -1 downto 0)
  );
end nvc_bug;

architecture Behavioral of nvc_bug is
  -- Attribute definitions
  type state_t is (idle, test_0, test_1, test_2, test_3);
  type state_array_t is array (0 to read_size - 1) of state_t;
  signal state_array : state_array_t;

begin

bug_generator: for i in 0 to write_size - 1 generate

    process (clk)
    begin
        if rising_edge(clk) then
            if(reset = '1')then
                state_array(i) <= idle;
                test_out(i) <= '0';
            else
                case state_array(i) is
                    when idle =>
                        if(write_en(i) = '1')then
                            state_array(i) <= test_0;
                            test_out(i) <= '1';
                        else
                            state_array(i) <= idle;
                            test_out(i) <= '0';
                        end if;
                    when test_0 =>
                        state_array(i) <= test_1;
                        test_out(i) <= '1';
                    when test_1 =>
                        state_array(i) <= test_2;
                        test_out(i) <= '0';
                    when test_2 =>
                        state_array(i) <= test_3;
                        test_out(i) <= '1';
                    when test_3 =>
                        state_array(i) <= idle;
                        test_out(i) <= '0';
                
                    when others =>
                        
                
                end case;
            end if;
        end if;
    end process;

end generate;

end Behavioral;
`

nickg, 2025-01-25:
Which version are you using? I get an error like this and no crash:

```
** Fatal: (init): index 2 outside of INTEGER range 0 to 1
    > test.vhd:30
    |
 30 |     bug_generator: for i in 0 to write_size - 1 generate
    |                        ^
```

alpkoyun, 2025-01-27:
1.16, It seems I have installed it directly by cloning the main branch.

nickg, 2025-01-29:
Could you also post the cocotb scripts you use to run it?

alpkoyun, 2025-01-30:
Hi, here is the Makefile and python script for the test. My original setup was using WSL Ubuntu on Windows with LLVM 14.0.6. I have changed my setup to an actual Ubuntu computer and still got the same crash. When vhdl source file, Makefile and the python file is in the same directory, calling command "make" should run the test.
```
TOPLEVEL_LANG ?= vhdl
PWD=$(shell pwd)
SRC_DIR := $(PWD)
VHDL_SOURCES = $(wildcard $(SRC_DIR)/*.vhd)
SIM:=nvc
TOPLEVEL := nvc_bug
MODULE   := test_nvc
include $(shell cocotb-config --makefiles)/Makefile.sim
```

python file named:test_nvc.py

```
import cocotb
from cocotb.triggers import Timer
from cocotb.clock import Clock

async def reset_sys(dut):
    dut.write_en.value = 0
    dut.reset.value = 1
    await Timer(200, units='ns')
    dut.reset.value = 0

@cocotb.test()
async def test_nvc(dut):
    clk=Clock(dut.clk,3,"ns")
    await cocotb.start(clk.start())
    await reset_sys(dut)
    await Timer(1, units='us')
```

nickg, 2025-02-02:
> My original setup was using WSL Ubuntu on Windows with LLVM 14.0.6. 

Thanks, I think LLVM 14 was the key thing here. Should be fixed now, can you test again?

commit 4c449ce9 (2025-02-02): Fix placement of LLVM unreachable instructions. Fixes #1145
commit 63c6af52 (2025-02-02): Fix placement of LLVM unreachable instructions. Fixes #1145
closed 2025-02-02 by nickg