nickg/nvc #1150: Expression coverage assertion crash + follow-up crash related to PSL

Blebowski, 2025-01-30:
Hi,

I have encountered another issue when porting large TB to NVC. When I enable expression coverage,
and build in the debug mode, I get crash during elaboration.

To reproduce:
```
git clone git@gitlab.fel.cvut.cz:canbus/ctucanfd_ip_core.git
cd ctucanfd_ip_core
git checkout b508c668f29d97f1fa980d1c6dbf333f8c578ea6
export CTU_TB_TOP_TARGET="tb_ctu_can_fd_rtl_vunit"
cd test
VUNIT_SIMULATOR=nvc ./run.py tb_rtl_test_fast_asic ctu_can_fd_tb.tb_top_ctu_can_fd.device_id
```

I get crash:
```
nvc: ../src/cov/cov-data.c:694: cover_add_items_for: Assertion `tree_kind(tree) == T_FCALL' failed.

*** Caught signal 6 (SIGABRT) ***

[0x5b868009f0c9] ../src/util.c:897 signal_handler
[0x76d3c024251f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x76d3c02969fc] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./nptl/pthread_kill.c:44 __pthread_kill_implementation
[0x76d3c02969fc] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./nptl/pthread_kill.c:78 __pthread_kill_internal
[0x76d3c02969fc] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./nptl/pthread_kill.c:89 pthread_kill@@GLIBC_2.34
[0x76d3c0242475] (/usr/lib/x86_64-linux-gnu/libc.so.6) ../sysdeps/posix/raise.c:26 raise
[0x76d3c02287f2] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./stdlib/abort.c:79 abort
[0x76d3c022871a] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./assert/assert.c:92 __assert_fail_base.cold
[0x76d3c0239e95] (/usr/lib/x86_64-linux-gnu/libc.so.6) ./assert/assert.c:101 __assert_fail
[0x5b868016d812] ../src/cov/cov-data.c:694 cover_add_items_for
[0x5b86801310bc] ../src/lower.c:1851 lower_logic_expr_coverage
[0x5b86801310bc] ../src/lower.c:2548 lower_fcall
[0x5b868011d01f] ../src/lower.c:5205 lower_expr
[0x5b868011fe1e] ../src/lower.c:12975 lower_rvalue
[0x5b868013783b] ../src/lower.c:5975 lower_var_assign
[0x5b868013783b] ../src/lower.c:7393 lower_stmt
[0x5b8680139abb] ../src/lower.c:6361 lower_sequence
[0x5b8680139abb] ../src/lower.c:11667 lower_process
[0x5b86800fc60f] ../src/elab.c:1948 elab_stmts
[0x5b86800fc4b4] ../src/elab.c:1462 elab_architecture
[0x5b86800fd6de] ../src/elab.c:1596 elab_stmts
[0x5b86800fc4b4] ../src/elab.c:1462 elab_architecture
[0x5b86800fd6de] ../src/elab.c:1596 elab_stmts
[0x5b86800fc4b4] ../src/elab.c:1462 elab_architecture
[0x5b86801b8440] ../src/rt/model.c:3791 call_with_model
[0x5b86800ff31e] ../src/elab.c:2261 elab
[0x5b8680097995] ../src/nvc.c:521 elaborate
[0x5b8680097995] ../src/nvc.c:2266 process_command
[0x5b868009a17e] ../src/nvc.c:2426 main
fail (P=0 S=0 F=1 T=1) ctu_can_fd_tb.tb_top_ctu_can_fd.device_id (0.8 s)
```

When I added hierarchy prints into the code, it seems to have problem at if condition expression in `src/can_top_level.vhd`:
```
    txtb_asr_gen : for i in 0 to txt_buffer_count - 1 generate
    begin

        process (txtb_state, pc_dbg.is_overload)
        begin
            if ((((txtb_state(i) = TXT_TRAN) or (txtb_state(i) = TXT_ABTP))) and
                  (pc_dbg.is_overload = '1')) then
                report "TXT Buffer should have been unlocked when node is in Overload frame!"
                severity error;
            end if;
        end process;

    end generate;
```
Though if I remove this process (just as a hack an try to continue), I get similar crash at e.g. following logic:
```
    bit_err_norm_valid <= '1' when (sp_control /= SECONDARY_SAMPLE and
                                    data_rx_synced /= data_tx and
                                    rx_trigger = '1' and
                                    bit_err_enable = '1')
                              else
                          '0';
```
in `src/bus_sampling/bit_err_detector.vhd`.

I tried to add prints just before the assertion failure, but I don't get the full log well printed.
When printing the `tree_kind` with the debug disabled (not to get the crash), I get `T_PROT_FCALL`,
so maybe it is as simple as extending the assertion for another allowed `tree_kind_t` in it.
I don't know.

When I try to extend the assertion to have `T_PROT_FCALL` allowed, I elaborate OK, but I crash at runtime
(due to different issue in a PSL cover directive):

```
        10000000 fs - default              -    INFO - Bit timing settings (Nominal):
        10000000 fs - default              -    INFO -   BRP: 2
        10000000 fs - default              -    INFO -   PH1: 7
        10000000 fs - default              -    INFO -   PROP: 5
        10000000 fs - default              -    INFO -   PH2: 7
        10000000 fs - default              -    INFO -   SJW: 5
        10000000 fs - default              -    INFO - 
        10000000 fs - default              -    INFO - Bit 
*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x73edbfbc9bac] ***

[0x5bb14b3990c9] ../src/util.c:897 signal_handler
[0x73ee7d64251f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x73edbfbc9bac] (/WORK/ctu_can_fd_clean/test/vunit_out/nvc/libraries/ctu_can_fd_tb/_CTU_CAN_FD_TB.TB_TOP_CTU_CAN_FD-TB.elab.251041.so) CTU_CAN_FD_TB.TB_TOP_CTU_CAN_FD.CTU_CAN_FD_VIP_INST.CTU_CAN_FD_VIP.FEATURE_TEST_AGENT_GEN.FEATURE_TEST_AGENT_INST.FEATURE_TEST_AGENT.TEST_NODE_INST.RX_BUFFER_INST.RX_BUF_COMMIT_AND_READ_COV
[0x5bb14b46e43e] ../src/jit/jit-core.c:692 jit_try_vcall
[0x5bb14b46ecc3] ../src/jit/jit-core.c:807 jit_vfastcall
[0x5bb14b4abb7c] ../src/rt/model.c:2478 update_property
[0x5bb14b4abb7c] ../src/rt/model.c:2744 async_update_property
[0x5bb14b4aa201] ../src/rt/model.c:402 deferq_run
[0x5bb14b4af4f3] ../src/rt/model.c:3326 model_cycle
[0x5bb14b4b1615] ../src/rt/model.c:3397 model_run
[0x5bb14b392dbe] ../src/nvc.c:956 run_cmd
[0x5bb14b392dbe] ../src/nvc.c:2268 process_command
[0x5bb14b391ac9] ../src/nvc.c:585 elaborate
[0x5bb14b391ac9] ../src/nvc.c:2266 process_command
[0x5bb14b39417e] ../src/nvc.c:2426 main
fail (P=0 S=0 F=1 T=1) ctu_can_fd_tb.tb_top_ctu_can_fd.device_id (1 min 25.8 s)
```

nickg, 2025-02-02:
Both of these should be fixed now, thanks.

closed 2025-02-02 by nickg

Blebowski, 2025-02-02:
Thanks for fixing this so quickly.