nickg/nvc #1152: Elaboration failure with hash of 00000000 expected

axman6, 2025-02-03:
I'm running into a similar issue to #769 but somehow something's depending on something with a hash of 00000000.  The commands being run by Cocotb are (with `...` replacing paths etc. that shouldn't be relevant, and names changed to protect the innocent):

```shell
nvc \
 --work=work \
 --std=08 \
 --map=unisim:.../unisim.08 \
 --map=XPM:.../xpm.08 \
 -M 128m \
 -H 128m \
 -a \
   .../SUPPORT.vhd \
   .../BAR.vhd \
   .../FOO.vhd \
   .../common/OTHER.vhd 
#    in directory .../sim_build

 nvc \
  --work=work \
  --std=08 \
  --map=unisim:.../unisim.08 \
  --map=XPM:.../xpm.08 \
  -M 128m \
  -H 128m \
  -e foo \
  --no-save \
  --jit \
  -gDATA_WIDTH=32 \
  -r \
  --load=.../site-packages/cocotb/libs/libcocotbvhpi_nvc.so 
#   in directory .../sim_build
```
which produces 
```
** Fatal: .../sim_build/work/WORK.FOO-BEHAVIORAL: design unit depends on 
          WORK.BAR with checksum 00000000 but the current version in the library has checksum 63edf8e7
** Note: this usually means WORK.FOO-BEHAVIORAL is outdated and needs to be reanalysed
```
This happens when I completely delete the `sim_build` directory. and re-run the test, which performs the above analysis and elaboration calls.

nickg, 2025-02-05:
This can happen if there is a circular dependency or a design unit depends on another one that was later overwritten. I've improved the error message here, can you try again with the latest master branch to see if that triggers? Failing that you'll need to try to come up with a complete minimal reproducer. E.g. by deleting stuff until the error no longer occurs.

axman6, 2025-02-06:
Thanks @nickg, yeah I believe this is related to a cyclic dependency we have which we're planning to remove. I'll a) try the latest master and b) see if removing the cyclic dep helps.

nickg, 2025-03-01:
As discussed above this is caused by a circular dependency which is not supported. There error message has been improved to make this clear.

closed 2025-03-01 by nickg