nickg/nvc #1153: t_item_range_value should have a range of integer'low to integer'high

JimLewis, 2025-02-04:
t_item_range_value should have a range of integer'low to integer'high

The valid range of a coverage bin is integer'low to integer'high.   I am getting many regression crashes in 1.15.1 (just loaded) due to a cast to t_item_range_value.

error message is
```
Error: AddBins(A, GenBin(-10, 10))
Error: ** Fatal: 0ms+0: value -10 outside of T_ITEM_RANGE_VALUE range 0 to 2147483647
Error:      > C:\OsvvmLibraries\osvvm\CoverageVendorApiPkg_NVC.vhd:143
Error:      |
Error:  143 |       ranges(i).min := t_item_range_value(bins(i).min);
Error:      |                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Error:    Procedure VENDORCOVBINADD [VENDORCOVHANDLETYPE, VENDORCOVRANGEARRAYTYPE, INTEGER, INTEGER, STRING] at C:\OsvvmLibraries\osvvm\CoverageVendorApiPkg_NVC.vhd:137
Error:    Procedure INSERTNEWBIN [COVERAGEIDTYPE, RANGEARRAYTYPE, INTEGER, INTEGER, INTEGER, INTEGER, STRING, REAL] at C:\OsvvmLibraries\osvvm\CoveragePkg.vhd:3046
Error:    Procedure INSERTBIN [COVERAGEIDTYPE, RANGEARRAYTYPE, INTEGER, INTEGER, INTEGER, INTEGER, STRING] at C:\OsvvmLibraries\osvvm\CoveragePkg.vhd:3101
Error:    Procedure ADDBINS [COVERAGEIDTYPE, STRING, INTEGER, INTEGER, COVBINTYPE] at C:\OsvvmLibraries\osvvm\CoveragePkg.vhd:3172
Error:    Procedure ADDBINS [COVERAGEIDTYPE, COVBINTYPE] at C:\OsvvmLibraries\osvvm\CoveragePkg.vhd:3216
Error:    Procedure ADDBINS [COVERAGEIDTYPE, COVBINTYPE] at C:\OsvvmLibraries\osvvm\CoveragePkg.vhd:8181
Error:    Process :negvaluecov:testproc at C:\SynthWorks\Dev\_osvvm\SandBox\NVC\Cov\NegValueCov.vhd:51
Error: child process exited abnormally
```

Simple VHDL Test Case reproducer:
```vhdl
library ieee ;
  use ieee.std_logic_1164.all ;
  use ieee.numeric_std_unsigned.all ;

library osvvm ; 
  context osvvm.OsvvmContext ; 

entity NegValueCov is
end NegValueCov ;
architecture test of NegValueCov is
begin
  TestProc : process
    variable A : CoverageIDType := NewID("Cov") ;
  begin
    SetTestName("NegValueCov") ; 
    
    print("AddBins(A, GenBin(-10, 10))") ;
    AddBins(A, GenBin(-10, 10)) ; 
    
    wait for 5 ns ; 
    
    print("AddBins(A, ALL_ILLEGAL)") ;
    AddBins(A, ALL_ILLEGAL) ; 
    
    AffirmIf(TRUE, "Previous ones Crash NVC if broken") ; 
    EndOfTestReports ; 

    std.env.stop ;
    wait ;
  end process ;
end test ;
```

The code around line 143 is:
```vhdl
  procedure VendorCovBinAdd(obj: inout VendorCovHandleType; bins: VendorCovRangeArrayType; Action: integer; atleast: integer; name: string )is
    variable item           : t_item_handle;
    variable ranges         : t_item_range_array(1 to bins'length);
    variable atleast_valid  : natural;
   begin
    for i in 1 to bins'length loop
      ranges(i).min := t_item_range_value(bins(i).min);
      ranges(i).max := t_item_range_value(bins(i).max);
    end loop;
```

BTW:  implementing coverage as a 64 bit number is something that people already want.

nickg, 2025-02-06:
Maybe I'm misunderstanding how this is used by I can't see why the range should be negative. Anyway I made it a signed 64-bit value.

closed 2025-02-06 by nickg

JimLewis, 2025-02-06:
Coverage range is a set of values that can describe anything the user decides it is.   So both positive and negative values can be possible.   As a result, OSVVM uses signed integer values.

JimLewis, 2025-02-06:
Does this update have a version number that I can check for?

nickg, 2025-02-06:
I'll include it in 1.15.2

commit 580250e6 (2025-02-06): Make t_item_range_value a signed 64-bit value. Fixes #1153
commit d06d576f (2025-02-06): Make t_item_range_value a signed 64-bit value. Fixes #1153