nickg/nvc #1155: invalid type kind T_ARRAY in range_of

axman6, 2025-02-06:
I'm getting the following crash when running quite a few of our tests using cocotb:

```shell
> nvc \
  --work=work \
  --std=08 \
  --map=unisim:/nix/store/1ip0gr0k3dxr8ng3jwip11k21x43r8lw-nvc-unisim-2022.2/unisim.08 \
  --map=XPM:/nix/store/rmk60wqlzb3k9qpyr7510fms7prxcsw9-nvc-xpm_vhdl-unknown/xpm.08 \
  -M 128m \
  -H 128m \
  --vhpi-debug \
  --vhpi-trace \
  -e muxnnwhole \
  --no-save \
  --jit \
  -gSOME_GENERIC1=9 \
  -gSOME_GENERIC2=128 \
  -gSOME_GENERIC3=1024 \
  -gSOME_GENERIC4=3 \
  -gSOME_GENERIC5=18 \
  -gSOME_GENERIC6=7 \
  -r \
  --ieee-warnings=off \
  --dump-arrays=64 \
  --vhpi-debug \
  --vhpi-trace-V \
  --wave=../../wave.ghw \
  -g \
  --load=/nix/store/bim3ncd8m0jvx2awqpi76sgq325akzcd-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so
** Fatal: (init): invalid type kind T_ARRAY in range_of
[0x4d1b8d] diag_femit
[0x40afa6] fatal_trace
[0x47cbaa] range_of
[0x491e82] lower_get_type_bounds
[0x4920be] lower_array_len
[0x49467c] lower_array_total_len
[0x49af93] lower_default_value.part.0
[0x49b5b6] lower_type_bounds_var
[0x4a5adb] lower_decls
[0x4a6be4] lower_pack_body
[0x48e4bc] unit_registry_get
[0x4e2f56] jit_fill_irbuf
[0x4f2922] jit_interp
[0x4f313f] jit_interp
[0x4e1f73] jit_try_vcall
[0x4e268d] jit_fastcall
[0x51ccbf] create_scope
[0x46e63e] elab_lower
[0x472dcc] elab_architecture
[0x51e335] call_with_model
[0x4734d5] elab
[0x412517] process_command
[0x40e7f0] main

nvc 1.16-devel (Using LLVM 19.1.6) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues
```
There's no context given so I'm not sure where to begin in tracking down where the problem might be in our code. It looks like similar problems have been reported in #953, #906, #842, #830, #750, and a few others. I can't seem to find any way to get any debugging info out of NVC to help me track down where things are going wrong so I don't know where to start in coming up with a test case. After spending a bit of time in LLDB trying to see if I could trace where in our code the error came from, the only things that seems potentially useful is the following definitions:

```vhdl
	type ArrayOfSigned is array (natural range <>) of signed;
	subtype ArrayOfSigned16 is ArrayOfSigned(open)(15 downto 0);
```

nickg, 2025-02-06:
If you go to the frame that's in `lower_decl` and do `call fmt_loc(stdout, &decl->object.loc)` it should print the location of the declaration the causes the issue.

axman6, 2025-02-10:
Thanks @nickg, I eventually managed to get that working*, and confirmed that the line of code it's trying to JIT is

```vhdl
        ype ArrayOfSigned is array (natural range <>) of signed;
	type InterlacedSignal is record
		Data : ArrayOfSigned;
		Valid : std_logic;
	end record;
	subtype Interlaced16bSignal is InterlacedSignal(Data(open)(15 downto 0));  -- This one
```
 This works in both GHDL and Vivado without issues. Any thoughts on what could be going wrong here?

\*why does using a debugger always feel like you're fighting it instead of it helping you? I ended up needing to resort to casting the address of fmt_loc into an appropriately typed function pointer.

axman6, 2025-02-10:
I'm not sure how relevant it is, but we were running into what appears to be another bug in nvc where it seems to believe that, given these definitions

```vhdl
	type ArrayOfSigned is array (natural range <>) of signed;
	subtype ArrayOfSigned16 is ArrayOfSigned(open)(15 downto 0);
	type InterlacedSignal is record
		Data : ArrayOfSigned;
		Valid : std_logic;
	end record;
	subtype Interlaced16bSignal is InterlacedSignal(Data(open)(15 downto 0));
```

```vhdl
entity SomeEntity is
	generic (
		IN_LANES : integer;
		IN_WIDTH : integer;
	);
	port (
		DataIn : in InterlacedSignal(Data(0 to IN_LANES - 1)(IN_WIDTH - 1 downto 0));
	);
end entity;
```
assigning to a port with type `InterlacedSignal` using

```vhdl
	signal SomeArr : ArrayOfSigned16(0 to IN_LANES - 1);
	signal DemodDataOutValid : std_logic;
...
		SOMETHING: entity SomeEntity
			generic map (
				IN_LANES => IN_LANES,
				IN_WIDTH => 16,
			)
			port map (
				DataIn.Data => SomeArr,
				DataIn.Valid => DemodDataOutValid,
			);
```
I run into this error. I managed to "fix" things by defining 
```vhdl
	function arr16_to_arr(x : ArrayOfSigned16) return ArrayOfSigned is
		variable y : ArrayOfSigned(x'range)(x(0)'length - 1 downto 0);
	begin
		for i in x'range loop
			y(i) := x(i);
		end loop;
		return y;
	end function;
```
and using 
```vhdl
	signal SomeArr : ArrayOfSigned16(0 to IN_LANES - 1);
	signal DemodDataOutValid : std_logic;
...
		SOMETHING: entity SomeEntity
			generic map (
				IN_LANES => IN_LANES,
				IN_WIDTH => 16,
			)
			port map (
				DataIn.Data => arr16_to_arr(SomeArr),
				DataIn.Valid => DemodDataOutValid,
			);
```
That function should be identity, given that `ArrayOfSigned16` is a subtype of `ArrayOfSigned`, as far as I understand VHDL's `subtype`: https://ics.uci.edu/~jmoorkan/vhdlref/subt_dec.html
> Since a subtype is the same type as its base type, assignments between subtype and base type onjects[sic] can be made without conversion:

I managed to track down one place where I could fix things, but we appear to have several other places that I can't track down and cause about half of our tests which pass in GHDL to fail under nvc.

nickg, 2025-02-11:
I managed to reproduce a similar crash. Can you try again with the latest master branch?

axman6, 2025-02-11:
Ok I've given this a go, and it now looks like it's helped some tests progress (to the point where tests which were failing to run are running to completion, but now with many warnings being printed - probably an "us" issue).

It appears it may have introduced another bug, which looks like a null pointer deref (at least at first glance, I can dive deeper if needed). This is in a different test than the one above, and looking at the call stack, this might be a different bug all together:

```
$ nvc \
 --work=work \
 --std=08 \
 --map=unisim:/nix/store/...-nvc-unisim-2022.2/unisim.08 \
 --map=XPM:/nix/store/...-nvc-xpm_vhdl-unknown/xpm.08 \
 -M 128m \
 -H 128m \
 -e testinstrument \
 --no-save \
 --jit \
 -r \
 --ieee-warnings=off \
 --dump-arrays=64 \
 --wave=../../wave.ghw \
 -g \
 --load=/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so

 exec(co, module.__dict__)

*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x50f609] ***

[0x415ca5] ../src/util.c:897 signal_handler
[0x7fffee682a6f] (/nix/store/...-glibc-2.40-36/lib/libc.so.6) 
[0x50f609] ../src/vhpi/vhpi-model.c:682 is_abstractDecl
[0x51310e] ../src/vhpi/vhpi-model.c:3930 build_anonymousSubTypeDecl
[0x5131be] ../src/vhpi/vhpi-model.c:4068 cached_typeDecl
[0x512e90] ../src/vhpi/vhpi-model.c:3862 build_subTypeDecl
[0x51311d] ../src/vhpi/vhpi-model.c:3933 build_anonymousSubTypeDecl
[0x5131be] ../src/vhpi/vhpi-model.c:4068 cached_typeDecl
[0x512c74] ../src/vhpi/vhpi-model.c:3738 build_arrayTypeDecl
[0x512e19] ../src/vhpi/vhpi-model.c:3850 build_subTypeDecl
[0x51311d] ../src/vhpi/vhpi-model.c:3933 build_anonymousSubTypeDecl
[0x5131be] ../src/vhpi/vhpi-model.c:4068 cached_typeDecl
[0x5134d8] ../src/vhpi/vhpi-model.c:1087 init_objDecl
[0x513786] ../src/vhpi/vhpi-model.c:4186 build_signalDecl
[0x513cd3] ../src/vhpi/vhpi-model.c:4596 vhpi_lazy_decls
[0x50f9c3] ../src/vhpi/vhpi-model.c:1319 expand_lazy_list
[0x514fee] ../src/vhpi/vhpi-model.c:2319 vhpi_handle_by_name
[0x7fffe47f441f] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) _ZN8VhpiImpl19native_check_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEP9GpiObjHdl
[0x7fffee209435] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libgpi.so) _ZL23gpi_get_handle_by_name_P9GpiObjHdlRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEP16GpiImplInterface
[0x7fffee209716] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libgpi.so) gpi_get_handle_by_name
[0x7fffee14973e] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/simulator.cpython-312-x86_64-linux-gnu.so) _ZL18get_handle_by_namePN12_GLOBAL__N_114gpi_hdl_ObjectIP9GpiObjHdlEEP7_object
[0x7fffe42de90d] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) method_vectorcall_VARARGS
[0x7fffe4367f6a] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyEval_EvalFrameDefault
[0x7fffe427d883] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyObject_VectorcallTstate.lto_priv.3
[0x7fffe42a497e] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _Py_slot_tp_getattr_hook
[0x7fffe4255d77] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) PyObject_GetAttr
[0x7fffe4365533] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyEval_EvalFrameDefault
[0x7fffe427d9db] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyObject_FastCallDictTstate
[0x7fffe427e344] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) slot_tp_init
[0x7fffe4219666] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) type_call
[0x7fffe427d710] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyObject_MakeTpCall
[0x7fffe4368117] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyEval_EvalFrameDefault
[0x7fffe43780e4] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) gen_send
[0x7fffe42ecd9d] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) method_vectorcall_O
[0x7fffe4367f6a] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyEval_EvalFrameDefault
[0x7fffe4291d3d] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) object_vacall
[0x7fffe4292994] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) PyObject_CallFunctionObjArgs
[0x7fffee1b7ba2] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotb.so) _embed_sim_init
[0x7fffee207df8] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libgpi.so) _Z14gpi_embed_initiPKPKc
[0x7fffe47f083b] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) _ZN16VhpiStartupCbHdl12run_callbackEv
[0x7fffe47f0edb] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) handle_vhpi_callback
[0x5109d2] ../src/vhpi/vhpi-model.c:1527 vhpi_global_cb
[0x510a69] ../src/vhpi/vhpi-model.c:4662 vhpi_run_callbacks
[0x511681] ../src/vhpi/vhpi-model.c:4723 vhpi_phase_cb
[0x4fd147] ../src/rt/model.c:450 global_event
[0x503793] ../src/rt/model.c:3394 model_run
[0x40f227] ../src/nvc.c:963 run_cmd
[0x40e1c7] ../src/nvc.c:2273 process_command
[0x40ec0d] ../src/nvc.c:589 elaborate
[0x40e1b8] ../src/nvc.c:2271 process_command
[0x410b6f] ../src/nvc.c:2431 main

nvc 1.16-devel (Using LLVM 19.1.6) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues
```

nickg, 2025-02-12:
> 
> It appears it may have introduced another bug, which looks like a null pointer deref (at least at first glance, I can dive deeper if needed). This is in a different test than the one above, and looking at the call stack, this might be a different bug all together - let me know if this needs a new ticket:
> 

That's a different bug. Please open another ticket, thanks.

closed 2025-02-12 by nickg