nickg/nvc #1159: JIT and OSVVM regressions

JimLewis, 2025-02-09:
I did the pull request, https://github.com/OSVVM/OSVVM-Scripts/pull/54, that adds --jit to OSVVM simulations.    It runs noticeably faster, however, I get one simulation crashing with:

```
Error: ** Fatal: 0ms+0: missing body for TBSCOREBOARDPKG.TBSB_TAG_INDEX_3_RAND-TAG_INDEX_3_RAND.TAGNAMETYPE$image
Error: ** Fatal: 0ms+0: missing body for TBSCOREBOARDPKG.TBSB_TAG_INDEX_3_RAND-TAG_INDEX_3_RAND.TAGNAMETYPE$image
Error: ** Fatal: 0ms+0: missing body for TBSCOREBOARDPKG.TBSB_TAG_INDEX_3_RAND-TAG_INDEX_3_RAND.TAGNAMETYPE$image
. . . 
. . .
Error: ** Fatal: 0ms+0: missing body for TBSCOREBOARDPKG.TBSB_TAG_INDEX_3_RAND-TAG_INDEX_3_RAND.TAGNAMETYPE$image
Error: ** Fatal: 0ms+0: missing body for TBSCOREBOARDPKG.TBSB_TAG_INDEX_3_RAND-TAG_INDEX_3_RAND.TAGNAMETYPE$image
Error: child process exited abnormally
```

Test case is a little too complicated to share here.   Email perhaps but it is not a public test case.

nickg, 2025-02-09:
Doing elaboration/run in a single step (see my latest OSVVM-Scripts PR) probably works around this, although it's certainly a real bug that needs to be fixed.

JimLewis, 2025-02-09:
Did a bisection search.   Here is a nice little reproducer:   Always interesting to find something simple like this in a fairly complex test that is causing the issue - NVC handled the hard stuff well.

```vhdl
entity TbSB_NVC_1 is 
end TbSB_NVC_1 ;

architecture test of TbSB_NVC_1 is 
  type TagNameType is (Fred, Betty, BamBam, Barney, Whilma, Pebbles, Dino) ;  
begin

  CtrlProc : process
  begin
    for tag in 0 to TagNameType'Pos(TagNameType'Right) loop 
      report "Tag: " & to_string(TagNameType'val(Tag))  ; 
    end loop ;

    std.env.stop ;
    wait ; 
  end process ; 

end test ; 
```

commit de292cdb (2025-03-25): Write JIT pack file using same method as native code gen

Fixes #1159
closed 2025-03-25 by nickg