nickg/nvc #1161: Null pointer dereference

axman6, 2025-02-12:
Following on from the discussion in #1155, we're now seeing the following crash, with what appears to be a null pointer dereference:

```
$ nvc \
 --work=work \
 --std=08 \
 --map=unisim:/nix/store/...-nvc-unisim-2022.2/unisim.08 \
 --map=XPM:/nix/store/...-nvc-xpm_vhdl-unknown/xpm.08 \
 -M 128m \
 -H 128m \
 -e testinstrument \
 --no-save \
 --jit \
 -r \
 --ieee-warnings=off \
 --dump-arrays=64 \
 --wave=../../wave.ghw \
 -g \
 --load=/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so

 exec(co, module.__dict__)

*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x50f609] ***

[0x415ca5] ../src/util.c:897 signal_handler
[0x7fffee682a6f] (/nix/store/...-glibc-2.40-36/lib/libc.so.6) 
[0x50f609] ../src/vhpi/vhpi-model.c:682 is_abstractDecl
[0x51310e] ../src/vhpi/vhpi-model.c:3930 build_anonymousSubTypeDecl
[0x5131be] ../src/vhpi/vhpi-model.c:4068 cached_typeDecl
[0x512e90] ../src/vhpi/vhpi-model.c:3862 build_subTypeDecl
[0x51311d] ../src/vhpi/vhpi-model.c:3933 build_anonymousSubTypeDecl
[0x5131be] ../src/vhpi/vhpi-model.c:4068 cached_typeDecl
[0x512c74] ../src/vhpi/vhpi-model.c:3738 build_arrayTypeDecl
[0x512e19] ../src/vhpi/vhpi-model.c:3850 build_subTypeDecl
[0x51311d] ../src/vhpi/vhpi-model.c:3933 build_anonymousSubTypeDecl
[0x5131be] ../src/vhpi/vhpi-model.c:4068 cached_typeDecl
[0x5134d8] ../src/vhpi/vhpi-model.c:1087 init_objDecl
[0x513786] ../src/vhpi/vhpi-model.c:4186 build_signalDecl
[0x513cd3] ../src/vhpi/vhpi-model.c:4596 vhpi_lazy_decls
[0x50f9c3] ../src/vhpi/vhpi-model.c:1319 expand_lazy_list
[0x514fee] ../src/vhpi/vhpi-model.c:2319 vhpi_handle_by_name
[0x7fffe47f441f] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) _ZN8VhpiImpl19native_check_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEP9GpiObjHdl
[0x7fffee209435] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libgpi.so) _ZL23gpi_get_handle_by_name_P9GpiObjHdlRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEP16GpiImplInterface
[0x7fffee209716] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libgpi.so) gpi_get_handle_by_name
[0x7fffee14973e] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/simulator.cpython-312-x86_64-linux-gnu.so) _ZL18get_handle_by_namePN12_GLOBAL__N_114gpi_hdl_ObjectIP9GpiObjHdlEEP7_object
[0x7fffe42de90d] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) method_vectorcall_VARARGS
[0x7fffe4367f6a] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyEval_EvalFrameDefault
[0x7fffe427d883] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyObject_VectorcallTstate.lto_priv.3
[0x7fffe42a497e] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _Py_slot_tp_getattr_hook
[0x7fffe4255d77] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) PyObject_GetAttr
[0x7fffe4365533] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyEval_EvalFrameDefault
[0x7fffe427d9db] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyObject_FastCallDictTstate
[0x7fffe427e344] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) slot_tp_init
[0x7fffe4219666] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) type_call
[0x7fffe427d710] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyObject_MakeTpCall
[0x7fffe4368117] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyEval_EvalFrameDefault
[0x7fffe43780e4] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) gen_send
[0x7fffe42ecd9d] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) method_vectorcall_O
[0x7fffe4367f6a] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) _PyEval_EvalFrameDefault
[0x7fffe4291d3d] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) object_vacall
[0x7fffe4292994] (/nix/store/...-python3-3.12.8/lib/libpython3.12.so.1.0) PyObject_CallFunctionObjArgs
[0x7fffee1b7ba2] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotb.so) _embed_sim_init
[0x7fffee207df8] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libgpi.so) _Z14gpi_embed_initiPKPKc
[0x7fffe47f083b] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) _ZN16VhpiStartupCbHdl12run_callbackEv
[0x7fffe47f0edb] (/nix/store/...-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) handle_vhpi_callback
[0x5109d2] ../src/vhpi/vhpi-model.c:1527 vhpi_global_cb
[0x510a69] ../src/vhpi/vhpi-model.c:4662 vhpi_run_callbacks
[0x511681] ../src/vhpi/vhpi-model.c:4723 vhpi_phase_cb
[0x4fd147] ../src/rt/model.c:450 global_event
[0x503793] ../src/rt/model.c:3394 model_run
[0x40f227] ../src/nvc.c:963 run_cmd
[0x40e1c7] ../src/nvc.c:2273 process_command
[0x40ec0d] ../src/nvc.c:589 elaborate
[0x40e1b8] ../src/nvc.c:2271 process_command
[0x410b6f] ../src/nvc.c:2431 main

nvc 1.16-devel (Using LLVM 19.1.6) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues

nickg, 2025-02-16:
Can you try again with the latest master branch?

axman6, 2025-02-17:
Thanks again @nickg for you quick work. With this change, I seem to make a little more progress, I have the following function which I was using to work around one of the previous issues:

```vhdl
	function arr16_to_arr(x : ArrayOfSigned16) return ArrayOfSigned is
		variable y : ArrayOfSigned(x'range)(x(0)'length - 1 downto 0);
	begin
		for i in x'range loop
			y(i) := x(i);
		end loop;
		return y;
	end function;
```

Which I used to convert an `ArrayOfSigned16(0 to IN_LANES - 1)` to `ArrayOfSigned(Data(0 to IN_LANES - 1)(IN_WIDTH - 1 downto 0))` (where IN_WIDTH is 16) on an in port of an entity with which uses similar generics to specify the size of it `InterlacedSignal`. When I use it, I get:

```VHDL
...
port DataIn : in InterlacedSignal(Data(0 to IN_LANES - 1)(IN_WIDTH - 1 downto 0));
...
port map (
    DataIn.Data => arr16_to_arr(DataOutArray),
    DataIn.Valid => DataOutValid,
)
```

```
Name       WORK.TESTINSTRUMENT.DUT.PM_CHANNELS(0).PM_CHANNEL.PLL.AUTO_ACQUISITION.DEINTERLACER.DATAIN.convert_in
Kind       function
Context    WORK.TESTINSTRUMENT.DUT.PM_CHANNELS(0).PM_CHANNEL.PLL.AUTO_ACQUISITION.DEINTERLACER
Blocks     1
Registers  14
Types      17
Variables  0
Result     #
Parameters 2
  r0    context                         // P<WORK.TESTINSTRUMENT.DUT.PM_CHANNELS(0).PM_CHANNEL.PLL.AUTO_ACQUISITION.DEINTERLACER>
  r1    cf                              // X<>
Begin
   0: r2 := var upref 1, DATAIN         // @<WORK.INTERLACINGSUPPORT.INTERLACEDSIGNAL${}>
      r3 := var upref 3, QDEMODDATAOUTARRAY // @<$<0..8>> => 0..8
      r4 := load indirect r3            // $<0..8> => 0..8
      r5 := resolved r4                 // @<0..8> => 0..8
      r6 := const 0                     // -2^31..2^31-1 => 0
      r7 := const 3                     // -2^31..2^31-1 => 3
      r8 := const 0                     // 0..1 => 0
      r9 := wrap r5 [r6 r7 r8]          // [*] : 0..8
      r10 := link package WORK.INTERLACINGSUPPORT // P<WORK.INTERLACINGSUPPORT>
      r11 := fcall WORK.INTERLACINGSUPPORT.ARR16_TO_ARR(39WORK.INTERLACINGSUPPORT.ARRAYOFSIGNED16)37WORK.INTERLACINGSUPPORT.ARRAYOFSIGNED r10, r9 // [*, *] : 0..8 => 0..8
      r12 := debug locus T_CONV_FUNC@0x7fffdc632eb0 // D<>
      r13 := record ref r2 field 0      // @<[*, *] : $<0..8>> => $<0..8>
      invalid := record ref r11 field 0  <----

-------------------------------------------------------------------- Captured stderr call ---------------------------------------------------------------------
** Fatal: (init): argument to record ref must be a pointer
[0x4bc82f] ../src/diag.c:1054 diag_femit
[0x4bc92d] ../src/diag.c:1078 diag_emit
[0x414ff3] ../src/util.c:610 fatal_trace
[0x4a9e91] ../src/vcode.c:5174 emit_record_ref
[0x48a03d] ../src/lower.c:932 lower_for_each_field_2
[0x499fb8] ../src/lower.c:11829 lower_converter
[0x49a130] ../src/lower.c:12184 lower_port_map
[0x49a771] ../src/lower.c:12600 lower_ports
[0x49df8f] ../src/lower.c:13227 lower_instance
[0x463889] ../src/elab.c:1343 elab_lower
[0x46786b] ../src/elab.c:1460 elab_architecture
[0x467e25] ../src/elab.c:1635 elab_instance
[0x466d52] ../src/elab.c:2023 elab_stmts
[0x467fcc] ../src/elab.c:2082 elab_block
[0x466db9] ../src/elab.c:2026 elab_stmts
[0x467883] ../src/elab.c:1462 elab_architecture
[... 85 lines cut ...]
[0x40fcc2] ../src/nvc.c:593 elaborate
[0x40f24e] ../src/nvc.c:2287 process_command
[0x412012] ../src/nvc.c:2451 main

nvc 1.16-devel (Using LLVM 19.1.6) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues
```

Interestingly, if I try to pull out the assignments to the DataIn ports into a single signal and then pass that in:

```vhdl
signal DataIn : InterlacedSignal(Data(0 to IN_LANES - 1)(15 downto 0));
...
DataIn.Data <= arr16_to_arr(QDemodDataOutArray);
-- DataInData <= QDemodDataOutArray; -- also causes the same error
DataIn.Valid <= DemodDataOutValid;
...
port map (
    DataIn => DataIn
)
```
I get the same null pointer dereference as above.

Please let me know if you need any more detail or if I can get you any more details from a debugger.

axman6, 2025-02-24:
I've updated to the latest master and tried the tests again. Interestingly, I'm seeing something strange related to cocotb here - when I run one of the failing tests using pytest, which executes the test via cocotb, I get crashes like the one above.

```
INFO: Running command nvc --work=work --std=08 --map=unisim:/nix/store/rx3c2bbvc7s61y31awmnz45xh0jyhiw5-nvc-unisim-2022.2/unisim.08 --map=XPM:/nix/store/8p3pghdcmpds8ffb3nww5rsy8dn2pzd6-nvc-xpm_vhdl-unknown/xpm.08 -M 128m -H 128m -e testinstrument --no-save --jit -r --ieee-warnings=off --dump-arrays=64 --wave=../../wave.ghw -g --load=/nix/store/bim3ncd8m0jvx2awqpi76sgq325akzcd-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so in directory /home/alex/code/instruments/tests/instruments/pid/sim_build/nvc
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:79   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VHPI registered
     0.00ns INFO     cocotb                             Running on nvc version 1.16-devel
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.2 from /nix/store/kbllc1rws6c7kqqbn1g8f2v9dm9a694p-python3-3.12.8-env/lib/python3.12/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1740358852
     0.00ns INFO     cocotb.regression                  Found test pid.test_pid
     0.00ns INFO     cocotb.regression                  running test_pid (1/1)
------------------------------------------------------------------------------------------------------------------------------------------------------------------- Captured stderr call --------------------------------------------------------------------------------------------------------------------------------------------------------------------
** Warning: the --ieee-warnings option may have no affect as the IEEE packages have already been initialised, pass --ieee-warnings as a global option instead
** Note: loading VHPI plugin /nix/store/bim3ncd8m0jvx2awqpi76sgq325akzcd-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so
/nix/store/kbllc1rws6c7kqqbn1g8f2v9dm9a694p-python3-3.12.8-env/lib/python3.12/site-packages/_pytest/assertion/rewrite.py:184: UserWarning: Python runners and associated APIs are an experimental feature and subject to change.
  exec(co, module.__dict__)

*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x5141c9] ***

[0x417ae6] ../src/util.c:908 signal_handler
[0x7fffee44ba6f] (/nix/store/h7zcxabfxa7v5xdna45y2hplj31ncf8a-glibc-2.40-36/lib/libc.so.6) 
[0x5141c9] ../src/vhpi/vhpi-model.c:683 is_abstractDecl
[0x517fab] ../src/vhpi/vhpi-model.c:4062 build_anonymousSubTypeDecl
[0x51805b] ../src/vhpi/vhpi-model.c:4200 cached_typeDecl
[0x517d0b] ../src/vhpi/vhpi-model.c:3988 build_subTypeDecl
[0x517fba] ../src/vhpi/vhpi-model.c:4065 build_anonymousSubTypeDecl
[0x51805b] ../src/vhpi/vhpi-model.c:4200 cached_typeDecl
[0x517aed] ../src/vhpi/vhpi-model.c:3864 build_arrayTypeDecl
[0x517c94] ../src/vhpi/vhpi-model.c:3976 build_subTypeDecl
[0x517fba] ../src/vhpi/vhpi-model.c:4065 build_anonymousSubTypeDecl
[0x51805b] ../src/vhpi/vhpi-model.c:4200 cached_typeDecl
[0x518375] ../src/vhpi/vhpi-model.c:1081 init_objDecl
[0x518623] ../src/vhpi/vhpi-model.c:4318 build_signalDecl
[0x518b77] ../src/vhpi/vhpi-model.c:4731 vhpi_lazy_decls
[0x51459f] ../src/vhpi/vhpi-model.c:1311 expand_lazy_list
[0x519e7c] ../src/vhpi/vhpi-model.c:2428 vhpi_handle_by_name
[0x7fffe461341f] (/nix/store/bim3ncd8m0jvx2awqpi76sgq325akzcd-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libcocotbvhpi_nvc.so) _ZN8VhpiImpl19native_check_createERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEP9GpiObjHdl
[0x7fffedf59435] (/nix/store/bim3ncd8m0jvx2awqpi76sgq325akzcd-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libgpi.so) _ZL23gpi_get_handle_by_name_P9GpiObjHdlRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEP16GpiImplInterface
[0x7fffedf59716] (/nix/store/bim3ncd8m0jvx2awqpi76sgq325akzcd-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/libs/libgpi.so) gpi_get_handle_by_name
[0x7fffede9973e] (/nix/store/bim3ncd8m0jvx2awqpi76sgq325akzcd-python3.12-cocotb-1.9.2/lib/python3.12/site-packages/cocotb/simulator.cpython-312-x86_64-linux-gnu.so) _ZL18get_handle_by_namePN12_GLOBAL__N_114gpi_hdl_ObjectIP9GpiObjHdlEEP7_object
[0x7fffe40fd90d] (/nix/store/qrc496n6fsqp4p5m5h8wmw5d5jwyw5mr-python3-3.12.8/lib/libpython3.12.so.1.0) method_vectorcall_VARARGS
[0x7fffe4186f6a] (/nix/store/qrc496n6fsqp4p5m5h8wmw5d5jwyw5mr-python3-3.12.8/lib/libpython3.12.so.1.0) _PyEval_EvalFrameDefault
[0x7fffe409c883] (/nix/store/qrc496n6fsqp4p5m5h8wmw5d5jwyw5mr-python3-3.12.8/lib/libpython3.12.so.1.0) _PyObject_VectorcallTstate.lto_priv.3
[0x7fffe40c397e] (/nix/store/qrc496n6fsqp4p5m5h8wmw5d5jwyw5mr-python3-3.12.8/lib/libpython3.12.so.1.0) _Py_slot_tp_getattr_hook
[0x7fffe4074d77] (/nix/store/qrc496n6fsqp4p5m5h8wmw5d5jwyw5mr-python3-3.12.8/lib/libpython3.12.so.1.0) PyObject_GetAttr
[0x7fffe4184533] (/nix/store/qrc496n6fsqp4p5m5h8wmw5d5jwyw5mr-python3-3.12.8/lib/libpython3.12.so.1.0) _PyEval_EvalFrameDefault
[... cut ...]
# seems to run to completion just fine
```

The test passes - there's warnings from our code, but the cocotb test runs to completion as far as I can tell. This means two things, 1) that I can't think of a good way to run a debugger to catch the null pointer exception above and 2) I guess cocotb is running the binary in a different way than the executable that's printed. @ktbarrett pining you on this one too since it seems to be cocotb related (and I'm not sure where to go with trying to help diagnose things further).

ktbarrett, 2025-02-24:
The stacktrace doesn't doesn't really help, you'll need to [get in there with a debugger](https://docs.cocotb.org/en/stable/building.html#envvar-COCOTB_ATTACH). Or ideally create a reproducer you can share.

axman6, 2025-02-24:
@ktbarrett Thanks for the link, that's exactly what I needed. I'll see how I go with that.

nickg, 2025-03-01:
Thanks, I managed to write a smaller test that triggers this. Can you test again with the latest master branch?

axman6, 2025-03-03:
@nickg Thanks, as always, for persisting with this one. I've run out tests and have a lot of them failing with the same output:

```
** Fatal: invalid type kind T_ARRAY in range_of
[0x4d2c9d] diag_femit
[0x40af84] fatal_trace
[0x47cefa] range_of
[0x4924d2] lower_get_type_bounds
[0x49270e] lower_array_len
[0x494d5c] lower_array_total_len
[0x49b823] lower_default_value.part.0
[0x49be46] lower_type_bounds_var
[0x4a695c] lower_decls
[0x4a7ac4] lower_pack_body
[0x48ea14] unit_registry_get
[0x4e3d48] jit_fill_irbuf
[0x4f36e2] jit_interp
[0x4f3eff] jit_interp
[0x4e2dfe] jit_try_vcall
[0x4e4a00] jit_try_call
[0x4d093d] eval_instance
[0x46f0d9] elab_lower
[0x472ef4] elab_architecture
[0x47390c] elab
[0x411c62] process_command
[0x40e7be] main
```

I'll look into whether I can see which code this refers to shortly and update the comment.

closed 2025-03-03 by nickg