nickg/nvc #1162: VHDL-2019 and FILE_PATH

JimLewis, 2025-02-13:
The function FILE_PATH returns the fully resolved absolute path of the current VHDL file as either type
LINE or STRING. The FILE_PATH does not include the FILE_NAME

Curerntly NVC is including the FILE_NAME in FILE_PATH.

JimLewis, 2025-02-13:
I am running nvc version NVC-1.15.1 on MSYS2 on Windows with:
```vhdl
use std.env.all ; 
    
entity test is 
end entity test ; 
Architecture T of test is
begin
  TestProc : process
  begin

    report "File: " & FILE_NAME & " Line: " & FILE_LINE & " Directory: " & FILE_PATH ; 

    wait ; 
  end process TestProc ; 
end architecture T ; 
```

This results in:
```
** Note: 0ms+0: File: Test.vhd Line: 10 Directory: C:\SynthWorks\Dev\_osvvm\SandBox\2019\File_Line_Path\Test.vhd
   Process :test:testproc at C:\SynthWorks\Dev\_osvvm\SandBox\2019\File_Line_Path\Test.vhd:52
```

FILE_PATH prints:   C:\SynthWorks\Dev\_osvvm\SandBox\2019\File_Line_Path\Test.vhd

It should print:  C:\SynthWorks\Dev\_osvvm\SandBox\2019\File_Line_Path\

nickg, 2025-02-20:
> Do we know the release number that fixes this?

I'll make a 1.15.2 release in the next week or so with the fix in.

commit 03fb2f33 (2025-02-16): Fix wrong FILE_PATH result with relative path. Fixes #1162
commit 3c9c4bfc (2025-02-16): Fix wrong FILE_PATH result with relative path. Fixes #1162