nickg/nvc #1169: Parallel generation of fst-files

oscargus, 2025-03-05:
Just a heads up that Verilator may introduce parallel writing of (multiple) fst-files: https://github.com/verilator/verilator/pull/5806

Currently there are no waveform viewers that can support this, but we hope to get support for Surfer soon.

Hence, it may be worthwhile to consider if this can be implemented in NVC. 

(Note that I do not think all the details are fully set yet, so very much a heads up.)

nickg, 2025-03-05:
BTW I don't think parallel writing is particularly useful to NVC at the moment as it doesn't do multi-threaded simulation.

oscargus, 2025-03-06:
> BTW I don't think parallel writing is particularly useful to NVC at the moment as it doesn't do multi-threaded simulation.

Ahh, then I didn't read that part of the docs good enough. Feel free to close unless you want to keep it as a motivation to get multi-threaded simulation going. :-)