nickg/nvc #1171: Unconstraint port mapping issue

yangyt96, 2025-03-12:
Hi, I have created a small example. This works in GHDL and Riviera but not in NVC:

```VHDL
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE ieee.numeric_std.ALL;

PACKAGE test_pkg IS
    TYPE t_unconstraint IS RECORD
        a : std_logic_vector;
        b : std_logic_vector;
    END RECORD;

    FUNCTION from_slv(x : std_logic_vector; dwidth : integer) RETURN t_unconstraint;
    FUNCTION to_slv(x   : t_unconstraint) RETURN std_logic_vector;
END PACKAGE;
PACKAGE BODY test_pkg IS
    FUNCTION from_slv(x : std_logic_vector; dwidth : integer) RETURN t_unconstraint IS
        VARIABLE tmp : t_unconstraint(
        a(dwidth - 1 DOWNTO 0),
        b(dwidth - 1 DOWNTO 0)
        );
    BEGIN
        tmp.a := x(dwidth - 1 DOWNTO 0);
        tmp.b := x(2 * dwidth - 1 DOWNTO dwidth);
        RETURN tmp;
    END FUNCTION;
    FUNCTION to_slv(x : t_unconstraint) RETURN std_logic_vector IS BEGIN
        RETURN x.b & x.a;
    END FUNCTION;
END PACKAGE BODY;

--------------------------------------------------------
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE ieee.numeric_std.ALL;

USE work.test_pkg.ALL;

ENTITY test IS
    PORT (
        ir_unconstraint : IN  t_unconstraint;
        or_unconstraint : OUT t_unconstraint
    );
END ENTITY test;
ARCHITECTURE rtl OF test IS
BEGIN
    or_unconstraint <= ir_unconstraint;
END ARCHITECTURE rtl;

--------------------------------------------------------
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE ieee.numeric_std.ALL;

ENTITY slv IS
    GENERIC (
        DWIDTH : integer
    );
    PORT (
        iv_data : IN  std_logic_vector(DWIDTH - 1 DOWNTO 0);
        ov_data : OUT std_logic_vector(DWIDTH - 1 DOWNTO 0)
    );
END ENTITY slv;
ARCHITECTURE rtl OF slv IS
BEGIN
    ov_data <= NOT iv_data;
END ARCHITECTURE rtl;

--------------------------------------------------------
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE ieee.numeric_std.ALL;

USE work.test_pkg.ALL;

ENTITY test_tb IS
END ENTITY test_tb;
ARCHITECTURE sim OF test_tb IS

    CONSTANT DWIDTH : integer := 6;

    SUBTYPE t_constraint IS t_unconstraint(
    a(DWIDTH - 1 DOWNTO 0),
    b(DWIDTH - 1 DOWNTO 0)
    );
    FUNCTION from_slv(x : std_logic_vector) RETURN t_constraint IS BEGIN
        RETURN from_slv(x, DWIDTH);
    END FUNCTION;

    SIGNAL test_to_slv : t_constraint;
    SIGNAL slv_to_test : std_logic_vector(DWIDTH - 1 DOWNTO 0);

BEGIN

    dut1 : ENTITY work.test
        PORT MAP(
            ir_unconstraint => from_slv(slv_to_test),
            or_unconstraint => test_to_slv
        );

    dut2 : ENTITY work.slv
        GENERIC MAP(DWIDTH => DWIDTH)
        PORT MAP(
            iv_data => to_slv(test_to_slv),
            ov_data => slv_to_test
        );

END ARCHITECTURE sim;
```

CLI:
```
nvc -a test.vhd
nvc -e test_tb
```

I get this error message:
````
Name       WORK.TEST_TB.DUT1
Kind       instance
Context    WORK.TEST_TB
Blocks     1
Registers  23
Types      23
  WORK.TEST_PKG.T_UNCONSTRAINT$         {[*] : $<0..8>, [*] : $<0..8>}
  WORK.TEST_PKG.T_UNCONSTRAINT          {[*] : 0..8, [*] : 0..8}
Variables  2
  IR_UNCONSTRAINT$cons                  // WORK.TEST_PKG.T_UNCONSTRAINT${}      
  IR_UNCONSTRAINT                       // WORK.TEST_PKG.T_UNCONSTRAINT${}, signal
Begin
   0: r0 := package init STD.STANDARD   // P<STD.STANDARD>
      r1 := package init WORK.TEST_PKG  // P<WORK.TEST_PKG>
      r2 := package init IEEE.STD_LOGIC_1164 // P<IEEE.STD_LOGIC_1164>
      r3 := package init IEEE.NUMERIC_STD // P<IEEE.NUMERIC_STD>
      r4 := var upref 1, TEST_TO_SLV    // @<WORK.TEST_PKG.T_UNCONSTRAINT${}>   
      r5 := index IR_UNCONSTRAINT$cons  // @<WORK.TEST_PKG.T_UNCONSTRAINT${}>   
      r6 := debug locus WORK.TEST+23    // D<>
      push scope locus r6
      r7 := null                        // @<WORK.TEST_PKG.T_UNCONSTRAINT{}>    
      r8 := index IR_UNCONSTRAINT       // @<WORK.TEST_PKG.T_UNCONSTRAINT${}>
      r9 := null                        // @<[*] : 0..8>
      r10 := record ref r8 field 0      // @<[*] : $<0..8>> => $<0..8>
      r11 := const 0                    // # => 0
      r12 := const 9                    // # => 9
      r13 := closure IEEE.STD_LOGIC_1164.RESOLVED(Y)U context r2 // C<0..8>     
      r14 := resolution wrapper r13 ileft r11 nlits r12 // R<0..8>
      r15 := const 1                    // # => 1
      r16 := link var r2 IEEE.STD_LOGIC_1164.STD_LOGIC_VECTOR // @<[*] : 0..8> => [*] : 0..8
      r17 := load indirect r16          // [*] : 0..8
      r18 := uarray len r17 dim 0       // #
      r19 := debug locus WORK.TEST_PKG+24 // D<>
      r20 := const 0                    // 0..8 => 0
      r21 := const 512                  // # => 512
      r22 := init signal count r18 size r15 value r20 flags r21 locus r19 offset
 r9 // $<0..8>
      resolve signal r22 resolution r14
      invalid := uarray left r22 dim 0   <----

** Fatal: cannot use uarray left with non-uarray type
[00007FF618D36EE0]
[00007FF618DE1406]
[00007FF618D35E73]
[00007FF618E88DD6] vhpi_is_printable+0x5eba6
[00007FF618DBC77C]
[00007FF618DBA790]
[00007FF618DCF707]
[00007FF618DCFA4E]
[00007FF618DDD4AC]
[00007FF618DD99A0]
[00007FF618D8BD87]
[00007FF618D92D3D]
[00007FF618D8D538]
[00007FF618D92D53]
[00007FF618D892B4]
[00007FF618D881BC]
[00007FF618D2E0D1]
[00007FF618D28D48]
[00007FF617CDD9B5]
[00007FF617CDDA06]
[00007FFEAC99259D] BaseThreadInitThunk+0x1d
[00007FFEAE90AF38] RtlUserThreadStart+0x28

Please report this bug at https://github.com/nickg/nvc/issues
````

yangyt96, 2025-03-13:
Btw, the software version is nvc 1.12.0.r1.ga2d80f74, using llvm 17.0.6 running on a Win11 machine

nickg, 2025-03-13:
This is fixed on the master branch now but I guess `iv_data` and `ov_data` should be `DWIDTH*2 - 1 downto 0`.

commit f5b3fa18 (2025-03-13): Fix crash with conversion function and unconstrained input port

Fixes #1171
closed 2025-03-13 by nickg