nickg/nvc #1175: unconstrained others error not covered by --relaxed option

weilanad, 2025-03-24:
**Description**
When assigning an others value to an unconstrained `std_logic_vector` the compiler throws an error. By using the `--relaxed` option this error should be suppressed.
With GHDL this error is suppressed by using `-frelaxed`.

**How to reproduce?**
In the following a minimal working example to reproduce 

```vhdl :file: test_unconstrained_others.vhdl
library IEEE;
use     IEEE.STD_LOGIC_1164.all;

entity module is
  port (
    vec : out std_logic_vector
  );
end entity;

architecture rtl of module is
begin
  vec <= (others => '1');
end architecture;
```
The following commands have been used for execution:

```sh 
nvc --std=08 -a --relaxed test_unconstrained_others.vhdl   # NVC
ghdl -a --std=08 -frelaxed test_unconstrained_others.vhdl  # GHDL
```

**Context**

- OS: Windows + MSYS2/UCRT64
- Origin:
  - [x] Built from sources: `nvc 1.15.2 (1.15.2.r0.g2fe12d4f)`

/cc @Paebbels

bpadalino, 2025-03-25:
@weilanad Not to say it's more correct, but ...
```vhdl
vec <= (vec'range => '1');
```
... works with NVC without the relaxed rules.

For unconstrained inputs, maybe this should be the recommended coding style?

Paebbels, 2025-03-25:
Is this a missing definition in the LRM? Because GHDL also has it behind `-frelaxed`, whereas Vivado and Riviera-PRO accept it as-is.

nickg, 2025-03-25:
In 9.3 it says

> The index range of an array aggregate that has an others choice shall be determinable from the context

and then

> e) As the value expression in an assignment statement, where the target is a declared object (or member thereof), and either the subtype of the target is a fully constrained array subtype or the target is a slice name.

But the target here is neither fully constrained nor a slice name so none of the listed contexts apply here and the `others` choice is illegal. See also #641.

Paebbels, 2025-03-25:
Is it possible, when unconstrained ports were allowed in VHDL, they forgot to modify this part?

nickg, 2025-03-25:
I think it would be reasonable to allow this in the same contexts where the 2019 "function knows vector size" feature works, and only if the `others` choice is the only choice.

BTW I noticed GHDL has an odd behaviour where it accepts the example above with `--std=08 -frelaxed` but not `-frelaxed --std=08`.

```
$ ghdl -a --std=08 -frelaxed test.vhd
$ ghdl -a -frelaxed --std=08 test.vhd
test.vhd:12:10:error: 'others' choice not allowed for an aggregate in this context
  vec <= (others => '1');
         ^
ghdl:error: compilation error
$ ghdl --version
GHDL 5.0.1 (4.1.0.r602.g37ad91899) [Dunoon edition]
```

Paebbels, 2025-03-25:
Are you going to report it?

nickg, 2025-03-25:
> Are you going to report it?

I'll leave it to you. Possibly it's intended that `--std` overrides `-frelaxed` (resets language settings?).

Paebbels, 2025-03-25:
@tgingold can you please have a look at https://github.com/nickg/nvc/issues/1175#issuecomment-2752295689
Is this intended behavior?

tgingold, 2025-03-25:
@Paebbels Yes `--std=XX` reset (or set) `-frelaxed`.

Paebbels, 2025-03-25:
@nickg I found it also in GHDL's documentation: https://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-std

I should have known that fact, but maybe knowledge like this get's lost once some scripts are written (like OSVVM-Scripts) ;).

JimLewis, 2025-03-25:
Certainly at elaboration time the port is fully constrained.  Or is there a requirement that it is fully constrained at compile time.

jobtijhuis, 2025-07-02:
I certainly would find it useful if the example given would work when using the `--relaxed` option. Of course you can refactor the code to use `'range` but since most synthesizers accept `vec <= (others => '1');` on unconstrained ports as-is we use this construct quite a lot.

nickg, 2025-07-18:
This is allowed with `--relaxed` now.

Paebbels, 2025-07-18:
Thanks. We'll test it next week.