nickg/nvc #1182: Regression for commit "Share model between elaboration and runtime"

avelure, 2025-04-04:
I noticed a recent regression that caused my previously working design to fail.
I have not managed to create a complete MWE that fails, but the crux of it is
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library propr_uvvm_vvc;
use propr_uvvm_vvc.uvvm_vvc_pkg.all; -- commenting this and it passes

package const is
  constant C_ADDRESS : unsigned(4 downto 0) := "00001";
end package const;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library uvvm_util;
context uvvm_util.uvvm_util_context;

library uvvm_vvc_framework;
use uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;

library propr_uvvm_vvc;
use propr_uvvm_vvc.vvc_context;

use work.const.all;

entity test is
end entity;

architecture beh of test is
  signal   address        : std_logic_vector(4 downto 0);
begin
i_ti_uvvm_engine : entity uvvm_vvc_framework.ti_uvvm_engine;

<---
instantiation of proprietary UVVM VVC, commenting this out fixes the issue
--->

  p_proc : process
  begin
    await_uvvm_initialization(VOID);
    assert C_ADDRESS = "00001" report "Address wrong 1 " & to_string(C_ADDRESS) severity failure;
    address        <= std_logic_vector(C_ADDRESS);
    assert C_ADDRESS = "00001" report "Address wrong 2 " & to_string(C_ADDRESS) severity failure;
    wait;
  end process;
end architecture;
```
The problem is that after the assignment the constant becomes "UUUUU". 
The issue disappears if 
- I change the constant to a std_logic_vector and remove the cast .
- Comment out the library reference to the uvvm vvc for the package
- Comment out the instantiation of the uvvm vvc

The constant and package name has no influence on the failure.

I bisected it down to the commit 6ddc89e7d06d7797dc98aa4bafd72820cbcdeaf8 though I see this is quite a big commit so maybe not so useful.

I can try to reduce the VVC or change it to a generic one at least if you don't have any idea what it could be already.

nickg, 2025-04-04:
Does it work if you elaborate and run in separate commands?

nickg, 2025-04-09:
Yeah I can't really see what could cause this. Maybe try incrementally taking things out of the VVC? Does it still fail if you comment out the `await_uvvm_initialization(VOID)` call? Another thing to try is setting a large heap size with e.g. `-H 1g` which will effectively disable garbage collection.

avelure, 2025-04-09:
`-H 1G` made no difference.
I managed to cook it down to only need the UVVM lib, but I guess it can be further reduced as well, but at least you can test now.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library uvvm_util;
context uvvm_util.uvvm_util_context;

package str is
  type t_cmd is
  record
    address : unsigned(4 downto 0);
  end record;
  function to_string(
    constant cmd : t_cmd
  ) return string;
end package str;
package body str is
  function to_string(
    constant cmd : t_cmd
  ) return string is
  begin
    return to_string(cmd.address, HEX_BIN_IF_INVALID, AS_IS, INCL_RADIX);
  end function;
end package body str;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.str.all;

package const is
  constant C_ADDRESS : unsigned(4 downto 0) := "00001";
end package const;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.const.all;

entity test is
end entity;

architecture beh of test is
  signal   address        : std_logic_vector(4 downto 0);
begin

  p_proc : process
  begin
    assert C_ADDRESS = "00001" report "Address 1 wrong " & to_string(C_ADDRESS) severity failure;
    address        <= std_logic_vector(C_ADDRESS);
    assert C_ADDRESS = "00001" report "Address 2 wrong " & to_string(C_ADDRESS) severity failure;
    wait;
  end process;
end architecture;
```
```
nvc --std=08 -LC:\path_to_uvvm_lib -a test_regression.vhd
nvc -LC:\path_to_uvvm_lib -e test -r
** Failure: 0ms+0: Address 2 wrong UUUUU
   Process :test:p_proc at C:\proj\public\minimal\test_regression.vhd:49
```

nickg, 2025-04-15:
Thanks for the test case. The problem is caused by of of the UVVM packages which has a signal declaration. Could you test again with the latest master branch?

avelure, 2025-04-21:
Yes, it works now, thank you :). 
I tested on 8fe8b209

closed 2025-04-21 by nickg