nickg/nvc #1192: All cocotb tests are failing

ktbarrett, 2025-04-18:
All cocotb tests currently fail on `master` with `Fatal: (init): cannot build VHPI typeDecl for T_ACCESS LINE`.

<details>
```
> make SIM=nvc TOPLEVEL_LANG=vhdl
rm -f results.xml
"make" -f Makefile results.xml
make[1]: Entering directory '/home/kaleb/dev/cocotb/tests/test_cases/test_cocotb'
# Make sure all libs in SOURCES_VHDL_* are mentioned in VHDL_LIB_ORDER and vice versa
\
/home/kaleb/.local/bin/nvc  --work=work:sim_build/work -L sim_build -a /home/kaleb/dev/cocotb/tests/test_cases/test_cocotb/../../../tests/designs/sample_module/sample_module_package.vhdl /home/kaleb/dev/cocotb/tests/test_cases/test_cocotb/../../../tests/designs/sample_module/sample_module_1.vhdl /home/kaleb/dev/cocotb/tests/test_cases/test_cocotb/../../../tests/designs/sample_module/sample_module.vhdl --preserve-case
rm -f results.xml
COCOTB_TESTCASE= COCOTB_TEST_FILTER= COCOTB_TEST_MODULES=" test_deprecated, test_synchronization_primitives, test_concurrency_primitives, test_tests, test_testfactory, test_timing_triggers, test_scheduler, test_clock, test_edge_triggers, test_async_coroutines, test_async_generators, test_handle, test_logging, pytest_assertion_rewriting, test_queues, test_sim_time_utils, test_start_soon, " COCOTB_TOPLEVEL=sample_module TOPLEVEL_LANG=vhdl \
   /home/kaleb/.local/bin/nvc  --work=work:sim_build/work -L sim_build \
  -e sample_module --no-save  \
  -r --load /home/kaleb/dev/cocotb/src/cocotb/libs/libcocotbvhpi_nvc.so
** Note: loading VHPI plugin /home/kaleb/dev/cocotb/src/cocotb/libs/libcocotbvhpi_nvc.so
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:94   in _embed_init_python              Using Python 3.12.3 interpreter at /home/kaleb/.pyenv/versions/system-venv/envs/cocotb/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:79   in gpi_print_registered_impl       VHPI registered
     0.00ns INFO     cocotb                             Running on nvc version 1.16-devel
** Fatal: (init): cannot build VHPI typeDecl for T_ACCESS LINE
[0x55fdf1985fae] ../src/util.c:612 fatal_trace
[0x55fdf1ae31d4] ../src/vhpi/vhpi-model.c:4190 vhpi_lazy_decls.lto_priv.0
[0x55fdf1ad965c] ../src/vhpi/vhpi-model.c:1311 expand_lazy_list
[0x55fdf1ad965c] ../src/vhpi/vhpi-model.c:1334 init_iterator
[0x55fdf1ade665] ../src/vhpi/vhpi-model.c:2514 vhpi_iterator
[0x7f5efc0b2a57] (/home/kaleb/dev/cocotb/src/cocotb/libs/libcocotbvhpi_nvc.so) src/cocotb/share/lib/vhpi/VhpiCbHdl.cpp:1045 VhpiIterator
[0x7f5efc0b2a57] (/home/kaleb/dev/cocotb/src/cocotb/libs/libcocotbvhpi_nvc.so) src/cocotb/share/lib/vhpi/VhpiCbHdl.cpp:971 _ZN8VhpiImpl14iterate_handleEP9GpiObjHdl18gpi_iterator_sel_e
[0x7f5efc208a54] (/home/kaleb/dev/cocotb/src/cocotb/simulator.cpython-312-x86_64-linux-gnu.so) src/cocotb/share/lib/simulator/simulatormodule.cpp:449 _ZL7iteratePN12_GLOBAL__N_114gpi_hdl_ObjectIP9GpiObjHdlEEP7_object
[0x7f5ef90e896a] (/usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0) ../Objects/descrobject.c:331 method_vectorcall_VARARGS
[0x7f5ef90dbafb] (/usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0) ../Include/internal/pycore_call.h:92 _PyObject_VectorcallTstate
[0x7f5ef90dbafb] (/usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0) ../Include/internal/pycore_call.h:325 PyObject_Vectorcall
[0x7f5ef90765ed] (/usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0) Python/bytecodes.c:2706 _PyEval_EvalFrameDefault
[0x7f5ef90dc36b] (/usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0) ../Include/internal/pycore_call.h:92 _PyObject_VectorcallTstate
[0x7f5ef90dc36b] (/usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0) ../Include/internal/pycore_call.h:562 _PyObject_CallFunctionVa
[0x7f5ef90dcafa] (/usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0) ../Objects/call.c:634 callmethod
[0x7f5ef90dcafa] (/usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0) ../Objects/call.c:653 PyObject_CallMethod
[0x7f5efc273f72] (/home/kaleb/dev/cocotb/src/cocotb/libs/libcocotb.so) src/cocotb/share/lib/embed/gpi_embed.cpp:267 _embed_sim_init
[0x7f5efc2a8b7c] (/home/kaleb/dev/cocotb/src/cocotb/libs/libgpi.so) src/cocotb/share/lib/gpi/GpiCommon.cpp:101 _Z14gpi_embed_initiPKPKc
[0x7f5efc0b2dbf] (/home/kaleb/dev/cocotb/src/cocotb/libs/libcocotbvhpi_nvc.so) src/cocotb/share/lib/vhpi/VhpiImpl.cpp:1093 _ZL16startup_callbackPv
[0x7f5efc0ae924] (/home/kaleb/dev/cocotb/src/cocotb/libs/libcocotbvhpi_nvc.so) src/cocotb/share/lib/vhpi/VhpiImpl.h:130 _ZN16VhpiStartupCbHdl3runEv
[0x7f5efc0af2db] (/home/kaleb/dev/cocotb/src/cocotb/libs/libcocotbvhpi_nvc.so) src/cocotb/share/lib/vhpi/VhpiCbHdl.cpp:33 _Z20handle_vhpi_callbackPK11vhpiCbDataS
[0x55fdf1ada31c] ../src/vhpi/vhpi-model.c:1519 vhpi_global_cb.lto_priv.0
[0x55fdf1ae3f5f] ../src/vhpi/vhpi-model.c:4799 vhpi_run_callbacks
[0x55fdf19a8d6a] ../src/rt/model.c:451 model_run
[0x55fdf19a8d6a] ../src/rt/model.c:984 run_cmd
[0x55fdf19a3916] ../src/nvc.c:2432 process_command
[0x55fdf19a7060] ../src/nvc.c:602 elaborate
[0x55fdf19a36f9] ../src/nvc.c:2430 process_command
[0x55fdf199fd4a] ../src/nvc.c:2600 main

nvc 1.16-devel (1.15.0.r168.g8fe8b209) (Using LLVM 18.1.3) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues

make[1]: *** [/home/kaleb/dev/cocotb/src/cocotb_tools/makefiles/simulators/Makefile.nvc:45: results.xml] Error 1
make[1]: Leaving directory '/home/kaleb/dev/cocotb/tests/test_cases/test_cocotb'
make: *** [/home/kaleb/dev/cocotb/src/cocotb_tools/makefiles/Makefile.inc:17: sim] Error 2
```
</details>

nickg, 2025-04-18:
I wasn't able to reproduce this: all the master branch cocotb tests pass for me. But anyway I think this crash should be fixed now.

closed 2025-04-18 by nickg

ktbarrett, 2025-04-18:
Just rechecked and it seems to be working now. Thanks!

commit 33c95878 (2025-04-18): Support access and file types in VHPI. Fixes #1192