nickg/nvc #1194: SEGFAULT with complex testbench with enabled expression coverage

tmeissner, 2025-04-22:
I have a fairly complex testbench, which produces a segfault when elaborated with nvc if expression coverage is enabled. All other coverage types are working, only expression coverage leads to the segfault. Sadly, my trys to reduce the testbench for a MWE weren't successful so far. My first thought was maybe another problem with generic packages and protected types, but also when commenting out these code parts, the segfault remains.

```
nvc --std=19 -L ./ -e --no-collapse --cover=statement,fsm-state,functional,branch,toggle,expression \
        --cover-spec=Coverage.sfile test

*** Caught signal 11 (SEGV_MAPERR) [address=0x74, ip=0x55b2900218f0] ***

[0x55b28ff5444d] ../src/util.c:906 signal_handler.lto_priv.0
[0x7f669ff98f8f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x55b2900218f0] ../src/cov/cov-data.c:641 cover_add_items_for
[0x55b28ffda291] ../src/lower.c:1853 lower_logic_expr_coverage
[0x55b28ffda291] ../src/lower.c:2425 lower_fcall.lto_priv.0
[0x55b28ffdea03] ../src/lower.c:5138 lower_expr.lto_priv.0
[0x55b28fff5417] ../src/lower.c:12808 lower_rvalue
[0x55b28ffd72b9] ../src/lower.c:1256 lower_subprogram_arg.lto_priv.0
[0x55b28ffda1cd] ../src/lower.c:2413 lower_fcall.lto_priv.0
[0x55b28ffdea03] ../src/lower.c:5138 lower_expr.lto_priv.0
[0x55b28fff5417] ../src/lower.c:12808 lower_rvalue
[0x55b28fff6597] ../src/lower.c:11832 lower_inertial_actual_process
[0x55b28fff4a6f] ../src/lower.c:13379 unit_registry_get
[0x55b29001b3ba] ../src/cgen.c:102 cgen_find_dependencies.lto_priv.0
[0x55b2900210b0] ../src/cgen.c:397 cgen
[0x55b28ff4f2c3] ../src/nvc.c:585 elaborate
[0x55b28ff4b853] ../src/nvc.c:2430 process_command
[0x55b28ff47f8b] ../src/nvc.c:2600 main

nvc 1.17-devel (1.16.0.r6.ge7b521fe) (Using LLVM 14.0.6) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues
```

Blebowski, 2025-04-23:
From the stack-trace this seems to be on some expressions with `std_logic` overloaded operands coverage.
If you uncoment `COVER_DEBUG_EMIT` macro in `src/cov/cov-data.c` and rebuild, you might be able to narrow
down the expression where it happends. Building with `--enable-debug` may help to fire some assertion too.

tmeissner, 2025-04-23:
The context looks like this:

```vhdl
EmmcSecureEraseP : process (Clk_i, Reset_n_i) is
  ...
begin
  if (not Reset_n_i) then
    ...
  elsif (rising_edge(Clk_i) then
    ...
      -- coverage off
      when others =>
        null;
      -- coverage on

    end case EmmcSecureEraseFsmC;
  end if;
end process EmmcSecureEraseP;
```

Another hint: the segfault happens during elaboration.

nickg, 2025-04-26:
Presumably the two crashes above also only happen with coverage enabled?

tmeissner, 2025-04-28:
Yes, only with coverage enabled and only with `statement` or `expression` coverage types.

nickg, 2025-04-29:
Can you try with `--enable-asan` in addition to `--enable-debug`? Probably also need to `export ASAN_OPTIONS=detect_leaks=0`.

tmeissner, 2025-04-30:
Strange: With your suggested configure parameters I get following error when compiling the first file. Even when the package declkarative part and body is empty ...

```
nvc --std=19 --work=crc -L ./ -a ../../../FpgaUtils.repo/Crc/Crc16MappingsP.vhd

=================================================================
==23==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 8200 byte(s) in 1 object(s) allocated from:
    #0 0x7f76d1f053b7 in __interceptor_calloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:77
    #1 0x560832de8a24 in xcalloc ../src/util.c:216
    #2 0x560832de8a24 in xcalloc_flex ../src/util.c:229
    #3 0x560832decc37 in get_table ../src/ident.c:173
    #4 0x560832decf82 in get_table ../src/ident.c:171
    #5 0x560832decf82 in ident_from_bytes ../src/ident.c:211
    #6 0x560832e97fbf in intern_strings ../src/common.c:1067
    #7 0x560832dd31d3 in main ../src/nvc.c:2478
    #8 0x7f76cb08e189  (/lib/x86_64-linux-gnu/libc.so.6+0x27189)

SUMMARY: AddressSanitizer: 8200 byte(s) leaked in 1 allocation(s).
```

```
$ nvc --version
nvc 1.17-devel (1.16.0.r10.g94cc9404) (Using LLVM 14.0.6) [debug]
```

nickg, 2025-04-30:
Did you also do `export ASAN_OPTIONS=detect_leaks=0` before running?

tmeissner, 2025-05-05:
Oh, I forgot to set `ASAN_OPTIONS` when running simulation. However, with that option, simulation runs fine without errors.

nickg, 2025-05-18:
@tmeissner can you test again with the latest master branch.

tmeissner, 2025-05-19:
@nickg With latest master branch it works now with all coverage types enabled. Thanks for the good work 🙂

closed 2025-05-19 by tmeissner