nickg/nvc #1200: Unit is too big to compile

jorkrohn, 2025-05-07:
After upgrading to version 1.16.0 from 1.15.2, one of the files in my test suite was not able to elaborate anymore, with the error message `unit <name> too big to compile`. This is a netlist that is provided from Xilinx as a simulation model for one of their IPs. 

I managed to generate a file, `large_entity.vhd`, that shows the same behavior. When the number of instances reaches ~11000, it fails in version 1.16.0. However, in version 1.15.2 it elaborated correctly up to ~33000 instances. 

I have attached `large_entity.vhd` and a python script that generates this file.

[large_entity.zip](https://github.com/user-attachments/files/20077871/large_entity.zip)

nickg, 2025-05-20:
Can you test again with the latest master branch?

nickg, 2025-05-21:
I think this should be fixed now (someone else reported the same crash in #1205) but it would be great if you have a simple way I can reproduce it.

nickg, 2025-05-22:
Try `export NVC_JIT_ASYNC=0 NVC_JIT_THRESHOLD=1` and see if that makes it easier to reproduce.

jorkrohn, 2025-05-22:
That made it fail during elaboration when instantiating the RAMB18SDP component from Xilinx/AMD:

```
*** Caught signal 11 (SEGV_MAPERR) [address=0x31e16a90, ip=0x7cbc31e16927] ***

[0x5a554ee38815] ../src/util.c:906 signal_handler.lto_priv.0
[0x7cbc3504532f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x7cbc31e16927] WORK.TEST_TB.I_RAM.RAMB18SDP.R1.RAMB18E1.SDP.WWB36.RAMB18E1_SDP_INST.RB18_INTERNAL_VHDL.GETWIDTH(I)I+0x27 [VHDL]

nvc 1.17-devel (1.16.0.r40.g108537b2) (Using LLVM 18.1.3) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues
```

Using the testbench file:

```
library ieee;
use ieee.std_logic_1164.all;

library unisim;
use unisim.vcomponents.all;

entity test_tb is
end entity test_tb;

architecture sim of test_tb is
begin

  i_ram : component RAMB18SDP
    port map (
      do      => open,
      dop     => open,
      di      => (others=>'0'),
      dip     => (others=>'0'),
      rdaddr  => (others=>'0'),
      rdclk   => '0',
      rden    => '0',
      regce   => '0',
      ssr     => '0',
      we      => (others=>'0'),
      wraddr  => (others=>'0'),
      wrclk   => '0',
      wren    => '0'
  );

end architecture sim;
```

I compiled and ran like this
```
nvc --work UNISIM -M 64m -a /tools/Xilinx/Vivado/2023.2/data/vhdl/src/unisims/unisim_retarget_VCOMP.vhd
nvc --work UNISIM -a /tools/Xilinx/Vivado/2023.2/data/vhdl/src/unisims/unisim_VPKG.vhd
nvc --work UNISIM -a /tools/Xilinx/Vivado/2023.2/data/vhdl/src/unisims/primitive/RAMB18E1.vhd
nvc --work UNISIM -a /tools/Xilinx/Vivado/2023.2/data/vhdl/src/unisims/retarget/RAMB18SDP.vhd
nvc -L . -a test_tb.vhd
nvc -L . -e test_tb
```

nickg, 2025-06-14:
The fix was a bit difficult to backport to the 1.16 branch. I'll make a 1.17 release in a month or so. In the meantime you could either continue using 1.15 or a snapshot of the master branch should be stable enough.