nickg/nvc #1204: View mode support VHDL 2019

xeblinho, 2025-05-20:
Add view mode support, currently returns error:
** Fatal: (init): cannot lower reference to T_VIEW_DECL

nickg, 2025-05-20:
Please provide a small test case that produces this error.

xeblinho, 2025-05-20:
interface.vhd:
`library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

use work.interface.all;

entity top is
  port (
    top_port : view if_view
  );
  end entity top;

  architecture logic of top is
  begin
    module_inst : entity work.module
      port map (
        module_port => top_port
      );
  
  end architecture logic;`

module.vhd:
`library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

use work.interface.all;

entity module is
  port (
    module_port : view if_view;
  );
  end entity module;`

top.vhd:
`library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

use work.interface.all;

entity top is
  port (
    top_port : view if_view
  );
  end entity top;

  architecture logic of top is
  begin
    module_inst : entity work.module
      port map (
        module_port => top_port
      );
  
  end architecture logic;`

commit 1ce17014 (2025-05-21): Handle unconnected views at top level. Fixes #1204
commit 664a515a (2025-06-04): Handle unconnected views at top level. Fixes #1204