nickg/nvc #1205: Fatal Error with current version

lemmuh, 2025-05-20:
Hi,

using my vhpi simulation setup with nvc (tag 1.16) works well.

With the current development version, very often (not always) the simulation stops with:

```
7ff1d843183c 90               nop
7ff1d843183d 90               nop
7ff1d843183e 90               nop
7ff1d843183f 90               nop
                              ;; skipping 32 zero bytes
7ff1d8431860 48b850f05af14956 movabs rax, 0x5649f15af050     ; __nvc_sched_waveform
                -0000
7ff1d843186a ffe0             jmp rax
--------------------------------------------------------------------------------
** Fatal: cannot handle relocation type 11 for symbol
** Note: setup:137ms run:87ms user:187ms sys:15ms maxrss:134456kB static:134kB
[0x5649f1491814] ../src/util.c:612 fatal_trace
[0x5649f15c95f3] ../src/jit/jit-code.c:1367 code_load_elf
[0x5649f15c95f3] ../src/jit/jit-code.c:1382 code_load_object
[0x5649f15c95f3] ../src/jit/jit-code.c:3438 jit_llvm_cgen.lto_priv.0
[0x5649f15893cd] ../src/thread.c:908 execute_task
[0x5649f1589580] ../src/thread.c:1068 steal_task
[0x5649f159049c] ../src/thread.c:1095 worker_thread
[0x5649f1590319] ../src/thread.c:571 thread_wrapper
[0x7ff1f5f317ea] (/usr/lib/libc.so.6) /usr/src/debug/glibc/glibc/nptl/pthread_create.c:448 start_thread
[0x7ff1f5fb518b] (/usr/lib/libc.so.6) ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78 __clone3

nvc 1.17-devel (1.16.0.r33.gf8bed76e) (Using LLVM 19.1.7) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues
```

Any ideas ?

nickg, 2025-05-20:
> 7ff1d843183e 90               nop

In the disassembly there should be one instruction marked with `<=======`. Can you paste that?

lemmuh, 2025-05-21:
Here the area:

```
7ff1b1060c54 e807050000       call 0x7ff1b1061160                                                                                                                                                                    7ff1b1060c59 498b8588000000   mov rax, qword ptr [r13 + 0x88]                                                                                                                                                        7ff1b1060c60 49638d90000000   movsxd rcx, dword ptr [r13 + 0x90]                                                                                                                                                     7ff1b1060c67 0fb6440808       movzx eax, byte ptr [rax + rcx + 8]                                                                                                                                                    7ff1b1060c6c 4883f803         cmp rax, 3                                                                                                                                                                             7ff1b1060c70 0f871e040000     ja 0x7ff1b1061094                                                                                                                                                                      7ff1b1060c76 ff24c500000000   jmp qword ptr [rax*8]               <=============                                                                                                                                     7ff1b1060c7d 498b4568         mov rax, qword ptr [r13 + 0x68]                                                                                                                                                        7ff1b1060c81 49634d70         movsxd rcx, dword ptr [r13 + 0x70]                                                                                                                                                     7ff1b1060c85 488903           mov qword ptr [rbx], rax                                                                                                                                                               7ff1b1060c88 48894b08         mov qword ptr [rbx + 8], rcx                                                                                                                                                           7ff1b1060c8c 48c7431001000000 mov qword ptr [rbx + 0x10], 1                                                                                                                                                          7ff1b1060c94 48896b18         mov qword ptr [rbx + 0x18], rbp                                                                                                                                                        7ff1b1060c98 48c7432000000000 mov qword ptr [rbx + 0x20], 0                                                                                                                                                          7
```

nickg, 2025-05-21:
Thanks. Can you try again with the current master branch?

lemmuh, 2025-05-21:
Seems to work again ! Thanks for your support.

closed 2025-05-21 by lemmuh