nickg/nvc #1225: NVC asks me to report this issue

yangyt96, 2025-06-30:
```
      r17 := const 1                    // # => 1
      put conversion r1 signal r16 count r17 values r15
      r18 := record ref r2 field 1      // @<*.*.*${}> => *.*.*${}
      invalid := record ref r12 field 1	 <----

** Fatal: invalid field 1
[0x55848aaa9545] ../src/diag.c:1054 diag_femit
[0x55848aaa9545] ../src/diag.c:1079 diag_emit
[0x55848a9c1a64] ../src/util.c:610 fatal_trace
[0x55848aa93448] ../src/vcode.c:5178 emit_record_ref
[0x55848aa674d6] ../src/lower.c:948 lower_for_each_field_2.lto_priv.0
[0x55848aa7f822] ../src/lower.c:11979 lower_converter
[0x55848aa82c7c] ../src/lower.c:12334 lower_port_map.lto_priv.0
[0x55848aa3f15d] ../src/lower.c:12758 lower_instance
[0x55848aa3f15d] ../src/lower.c:1335 elab_lower.lto_priv.0
[0x55848aa40aa1] ../src/elab.c:1452 elab_architecture.lto_priv.0
[0x55848aa47fd7] ../src/elab.c:1627 elab_stmts.lto_priv.0
[0x55848aa40ab7] ../src/elab.c:1454 elab_architecture.lto_priv.0
[0x55848a9e0c02] ../src/elab.c:2229 elab
[0x55848a9e0c02] ../src/elab.c:511 elaborate
[0x55848a9dd25c] ../src/nvc.c:2210 process_command
[0x55848a9d8ebb] ../src/nvc.c:2370 main

nvc 1.15.2 (Using LLVM 14.0.6) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues

fail (P=0 S=0 F=2 T=59) *.*.*.*(0.4 s)
```

nickg, 2025-07-03:
Are you able to provide a small example that triggers this crash or try on the most recent release?

yangyt96, 2025-07-06:
Hi @nickg, I need quite a lot of time to figure out a minimal example for this error. If u have time to wait please don't close this issue.

yangyt96, 2025-07-22:
> Are you able to provide a small example that triggers this crash or try on the most recent release?

Hi @nickg , I have successfully to create a minimal example to generate this bug. Please check the 1st comment

nickg, 2025-07-22:
Can you test with the latest 1.17.0 release? I tried the example above but it doesn't crash for me.