nickg/nvc #1252: Issue with `install-quartus.sh`: Missing `fiftyfivenm_atoms.vhd` in GitHub Action Environment

nselvara, 2025-07-28:
Hi there,

I'm currently setting up a GitHub Action workflow that integrates NVC and the Quartus simulation libraries.

To recreate the Quartus library environment in the CI container, I cloned [asicguy/gplgpu](https://github.com/asicguy/gplgpu), which includes a copy of the Quartus simulation libraries under `hdl/sim_lib/`. I then expose this path via the `QUARTUS_ROOTDIR` environment variable and attempt to install the Quartus libraries via:

```bash
nvc --install quartus
````

However, the script fails because it expects `fiftyfivenm_atoms.vhd`, which doesn't exist in my copy of the libraries:

```
** Fatal: opening /opt/intelFPGA/20.1/quartus/eda/sim_lib/fiftyfivenm_atoms.vhd: No such file or directory
```

Locally, I also don't have the `fiftyfivenm` variant, instead, I have libraries for newer process nodes (e.g., `20nm`, `maxv`, etc.).

### Workflow Snippet:

```yaml
- uses: nickg/setup-nvc@v1
  with:
    version: latest
- run: |
    git clone https://github.com/nselvara/gplgpu
    cd gplgpu
    mkdir -p /opt/intelFPGA/20.1/quartus/eda/
    cp -r ./hdl/sim_lib/ /opt/intelFPGA/20.1/quartus/eda/

    ls /opt/intelFPGA/20.1/quartus/eda/sim_lib/

    nvc --version
    nvc --install xpm_vhdl
    nvc --install quartus
  env:
    QUARTUS_ROOTDIR: "/opt/intelFPGA/20.1/quartus"
```

### Questions:

- Is there a supported way to override or skip missing device families during `--install quartus`?
- Could the install script be updated to detect which libraries are available instead of assuming a specific one like `fiftyfivenm`?
- Would you recommend an alternative strategy for CI integration without relying on unavailable libraries?

Thanks in advance, really appreciate the work behind NVC!

nickg, 2025-07-29:
I adjusted the script so it checks if those file are present. For the current release you could create empty files before running the script (e.g. `touch $QUARTUS_ROOTDIR/eda/sim_lib/fiftyfivenm_atoms.vhd`).

nselvara, 2025-07-29:
Wow, that was fast. Thank you very much for the quick fix. I will definitely do so, as you suggested.

commit ee1cfdcd (2025-07-30): Check for missing files in the Quartus install script

Fixes #1252