nickg/nvc #1253: Crash with unconstrained array elements of different sizes

NikLeberg, 2025-07-28:
I managed to crash NVC with the following:

```vhd
entity ent is end entity;

architecture arch of ent is
    type matrix_t is array (natural range <>) of bit_vector;
    signal x : bit_vector(0 to 1);
    signal y : bit_vector(0 to 2);
begin
    b0 : block is
        port (m : in matrix_t(0 to 1));
        port map (m(0) => x, m(1) => y);
    begin
    end block;
end architecture;
```

A constrained array with unconstrained elements is assigned element-wise in a port map.

```
# nvc -a unconstrained.vhd -e ent

Name       WORK.ENT.B0
Kind       instance
Context    WORK.ENT
Blocks     1
Registers  13
Types      12
Variables  1
  M                                     // [*, *] : $<0..1>, signal
Begin
   0: r0 := package init STD.STANDARD   // P<STD.STANDARD>
      r1 := var upref 1, X              // @<$<0..1>> => 0..1
      r2 := load indirect r1            // $<0..1> => 0..1
      r3 := var upref 1, Y              // @<$<0..1>> => 0..1
      r4 := load indirect r3            // $<0..1> => 0..1
      r5 := const 0                     // -2^31..2^31-1 => 0
      r6 := const 1                     // -2^31..2^31-1 => 1
      r7 := const 0                     // 0..1 => 0
      r8 := const 1                     // 0..1 => 1
      r9 := null                        // @<0..1>
      r10 := wrap r9 [r5 r6 r7]         // [*] : 0..1 => @<0..1>
      r11 := const 1                    // # => 1
      r12 := const 2                    // # => 2        <----

** Fatal: (init): array dimension 1 out of bounds
[0x5da55ffb627a] ../src/util.c:612 fatal_trace
[0x5da56008d2c2] ../src/vcode.c:4683 emit_uarray_len
[0x5da56005fcd2] ../src/lower.c:523 lower_array_stride.lto_priv.0
[0x5da560063e9d] ../src/lower.c:568 lower_array_total_len.lto_priv.0
[0x5da56007764d] ../src/lower.c:7768 lower_sub_signals.lto_priv.0
[0x5da5600805a2] ../src/lower.c:11862 lower_port_signal.lto_priv.0
[0x5da5600399fd] ../src/lower.c:12093 lower_instance
[0x5da5600399fd] ../src/lower.c:1420 elab_lower.lto_priv.0
[0x5da5600440a2] ../src/elab.c:2219 elab_stmts.lto_priv.0
[0x5da56003b05c] ../src/elab.c:1634 elab_architecture.lto_priv.0
[0x5da55ffd9259] ../src/elab.c:2341 elab
[0x5da55ffd9259] ../src/elab.c:553 elaborate
[0x5da55ffd5b8e] ../src/nvc.c:2443 process_command
[0x5da55ffd6f3b] ../src/nvc.c:2056 cover_merge_cmd
[0x5da55ffd6f3b] ../src/nvc.c:2469 process_command
[0x5da55ffd1679] ../src/nvc.c:2607 main

nvc 1.17.1 (Using LLVM 14.0.6) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues
```

Note that with a port map of `port map (m => (x, y));` NVC prints:
```# nvc -a unconstrained.vhd
** Error: length of value 3 does not match length of target 2
    > unconstrained.vhd:10
    |
 10 |         port map (m => (x, y));
    |                            ^
```

Which makes totally sense, because sub elements of an array are expected/required to all have the same length. Which is not the case here.

Cheers, Nik

NikLeberg, 2025-09-03:
Hey, this just in: It actually crashes even when both assignments have the same size. I.e. it seems that this is not supported.

I tried some permutations. Maybe the following helps to narrow it down. What works is, if the port is mapped to a fully constrained signal:
```vhd
entity ent is end entity;

architecture arch of ent is
    type matrix_t is array (natural range <>) of bit_vector;
    signal x : bit_vector(0 to 1) := "01";
    signal y : bit_vector(0 to 1) := "10";
    signal n : matrix_t(0 to 1)(0 to 1);
begin
    n <= (x, y);

    b0 : block is
        port (m : in matrix_t(0 to 1));
        port map (m => n);
    begin
        p0 : process is
        begin
            wait for 1 ns;
            assert m(0) = "01";
            assert m(1) = "10";
            wait;
        end process;
    end block;
end architecture;
```
This works.

But if the array aggregate is constructed inside the port map i.e. `port map (m => (x, y));` then it crashes on an assertion:
```
# nvc -a ../test/bounds/issue1253.vhd -e ent -r
nvc: ../src/lower.c:11626: lower_constrain_port: Assertion `tree_kind(value) == T_CONV_FUNC' failed.

*** Caught signal 6 (SIGABRT) ***

[0x57a3ccec8713] ../src/util.c:906 signal_handler
-->    show_stacktrace();
[0x710ac97ac32f] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x710ac9805b2c] (/usr/lib/x86_64-linux-gnu/libc.so.6) pthread_kill
[0x710ac97ac27d] (/usr/lib/x86_64-linux-gnu/libc.so.6) raise
[0x710ac978f8fe] (/usr/lib/x86_64-linux-gnu/libc.so.6) abort
[0x710ac978f81a] (/usr/lib/x86_64-linux-gnu/libc.so.6) 
[0x710ac97a2516] (/usr/lib/x86_64-linux-gnu/libc.so.6) __assert_fail
[0x57a3ccf8841e] ../src/lower.c:11626 lower_constrain_port
             tree_t value = tree_value(map);
-->          assert(tree_kind(value) == T_CONV_FUNC);
[0x57a3ccf88e21] ../src/lower.c:11797 lower_ports
          if (type_is_unconstrained(type))
-->          bounds_reg = lower_constrain_port(lu, port, i, block, map_regs);
          else if (!type_const_bounds(type))
[0x57a3ccf8b273] ../src/lower.c:12497 lower_instance
       lower_generics(lu, block, primary);
-->    lower_ports(lu, ds, block);
       lower_decls(lu, block);
[0x57a3ccf397b4] ../src/elab.c:1254 elab_lower
       else
-->       ctx->lowered = lower_instance(ctx->registry, ctx->parent->lowered,
                                        elab_driver_set(ctx), ctx->cover, b);
[0x57a3ccf3d792] ../src/elab.c:2298 elab_block
       if (elab_new_errors(&new_ctx) == 0) {
-->       elab_lower(b, &new_ctx);
          elab_stmts(t, &new_ctx);
[0x57a3ccf3d503] ../src/elab.c:2250 elab_stmts
          case T_BLOCK:
-->          elab_block(s, ctx);
             break;
[0x57a3ccf3b8bc] ../src/elab.c:1763 elab_architecture
          elab_stmts(entity, &new_ctx);
[... 105 lines cut ...]
       else
[0x5ff7046e3022] ../src/nvc.c:567 elaborate
-->    tree_t top = elab(obj, state->jit, state->registry, state->mir, cover,
                         NULL, state->model);
[0x5ff7046e7714] ../src/nvc.c:2464 process_command
       case 'e':
-->       return elaborate(argc, argv, state);
       case 'r':
[0x5ff7046e2571] ../src/nvc.c:311 analyse
-->    return argc > 1 ? process_command(argc, argv, state) : EXIT_SUCCESS;
    }
[0x5ff7046e76fa] ../src/nvc.c:2462 process_command
       case 'a':
-->       return analyse(argc, argv, state);
       case 'e':
[0x5ff7046e7c9e] ../src/nvc.c:2628 main
-->    const int ret = process_command(argc, argv, &state);
```

But sure, if the port is fully constrained `port (m : in matrix_t(0 to 1)(0 to 1));` then everything works as expected.

NVC is latest master (`git describe HEAD > r1.15.0-445-gac271cbd`)

As this seems to not be supported, shall I go and implement a simple _sorry, I can't do that dave_?
Alternatively I did already work on an detection of the initial issue (_constrained array with unconstrained and missmatched elements is assigned element-wise_) such that it would detect that as an error.

NikLeberg, 2025-09-03:
In other news, GHDL also crashes with everything I throw at it. QuestaSim (v21.1) accepts and simulates everything though.

nickg, 2025-09-03:
> But if the array aggregate is constructed inside the port map i.e. port map (m => (x, y)); then it crashes on an assertion:

I just fixed this crash but it's a bit different to the original issue. The problem is that there is an anonymous signal created that gets assigned `(x, y)` and is then used to constrain the port. You should get the same crash if you write `m => intertial n`.

nickg, 2025-09-07:
I think all these cases should be fixed now.

closed 2025-09-07 by nickg