nickg/nvc #1255: Integer literal -9223372036854775808 is not accepted in VHDL-2019

Paebbels, 2025-07-29:
VHDL-2019 increased the range of `integer` to a signed 64-bit integer. In addition, the bounds were changed to truly implement a 2s-complement range.

```vhdl
type integer is range -9223372036854775808 to 9223372036854775807;
```

Error message:
```
** Error: value 9223372036854775808 is outside implementation defined range of universal_integer
```
It looks like the lexer parses the number as a positive number, but not as a negative number.  
I have seen similar lexer behavior in Vivado.

The following testcase checks it and fails for `nvc --std=19 -a ..\..\Ranges.2019.vhdl`:
It might fail on unit `Hz` vs. `hz`. I need to check the LRM for it.
```vhdl
entity Ranges is
	port (
		signal Clock : in  bit;
		signal LED   : out bit
	);
end entity;

architecture test of Ranges is
	type int64 is range -9223372036854775808 to 9223372036854775807;

	type freq is range natural'low to natural'high units
		Hz;
		kHz = 1000 Hz;
		MHz = 1000 kHz;
		GHz = 1000 MHz;
		THz = 1000 GHz;
	end units;
	
	function contains(str : string; find : character) return boolean is
	begin
		for i in str'range loop
			return true when str(i) = find;
		end loop;
		return false;
	end function;
	
	constant freq_5Hz      : string := freq'image(5 Hz);
	constant freq_5Hz_unit : string := freq_5Hz(freq_5Hz'high - 1 to freq_5Hz'high);
begin
	-- IMPORTANT: activate assertions in synthesis settings
	
	assert integer'low   <= -9223372036854775808    report "FAILED: integer'low   <= -9223372036854775808   ; integer'low   = " & integer'image(integer'low)    ; -- & " / " & to_image(integer'low);
	assert integer'high  >=  9223372036854775807    report "FAILED: integer'high  >=  9223372036854775807   ; integer'high  = " & integer'image(integer'high)   ; -- & " / " & to_image(integer'high);
	assert natural'low    =           0             report "FAILED: natural'low    =                    0   ; natural'low   = " & natural'image(natural'low)    ; -- & " / " & to_image(natural'low);
	assert natural'high  >=  9223372036854775807    report "FAILED: natural'high  >=  9223372036854775807   ; natural'high  = " & natural'image(natural'high)   ; -- & " / " & to_image(natural'high);
	assert positive'low   =           1             report "FAILED: positive'low   =                    1   ; positive'low  = " & positive'image(positive'low)  ; -- & " / " & to_image(positive'low);
	assert positive'high >=  9223372036854775807    report "FAILED: positive'high >=  9223372036854775807   ; positive'high = " & positive'image(positive'high) ; -- & " / " & to_image(positive'high);

	assert int64'low     <= -9223372036854775808    report "FAILED: int64'low     <= -9223372036854775808   ; int64'low     = " & int64'image(int64'low)    ; -- & " / " & to_image(int64'low);
	assert int64'high    >=  9223372036854775807    report "FAILED: int64'high    >=  9223372036854775807   ; int64'high    = " & int64'image(int64'high)   ; -- & " / " & to_image(int64'high);
											  
	assert time'low      <= -9223372036854775807 fs report "FAILED: time'low      <= -9223372036854775807 fs; time'low      = " & time'image(time'low)  ; -- & " / " & to_image(time'low);
	assert time'high     >=  9223372036854775807 fs report "FAILED: time'high     >=  9223372036854775807 fs; time'high     = " & time'image(time'high) ; -- & " / " & to_image(time'high);

	assert freq'low       =                    0 Hz report "FAILED: freq'low      <=                    0 Hz; freq'low      = " & freq'image(freq'low)  ; -- & " / " & to_image(freq'low);
	assert freq'high      =  9223372036854775807 Hz report "FAILED: freq'high     >=  9223372036854775807 Hz; freq'high     = " & freq'image(freq'high) ; -- & " / " & to_image(freq'high);
	
	assert freq_5Hz_unit = "Hz"                     report "FAILED: Unit of '5 Hz' is 'Hz'; Unit = " & freq_5Hz_unit;
	assert not contains(freq_5Hz, '.')              report "FAILED: Physical type is an integer number";
	
	LED <= not LED when rising_edge(Clock);
end architecture;
```

Source: https://github.com/Paebbels/VHDL-Tool-Checks  
(This code was published for bug reporting at AMD.)

nickg, 2025-07-29:
> It looks like the lexer parses the number as a positive number, but not as a negative number.

There are no negative integer literals in VHDL. An `abstract_literal` is always positive or zero (see 15.5). The expression `-123` is parsed like `- ( 123 )` and I don't believe the 2019 revision changes this. In your case 9223372036854775808 (2^63) is outside the minimum range that the standard requires for `universal_integer`. The simple fix for this is to write `-9223372036854775807 - 1` instead (e.g. see the definition of `INTEGER` in `lib/std.19/standard.vhd`).

Paebbels, 2025-07-29:
> There are no negative integer literals in VHDL.
I suspected this, because I saw a similar behavior at `-3 mod 2` vs. `(-3) mod 2`.

Was this the reason why VHDL before 2019 used -2^31 - 1 as the lower bound instead of -2^31?

nickg, 2025-07-30:
> 
> Any thought on `hz` vs. `Hz`?

It should be `"hz"` in lower case. From the definition of `'image` in 16.2.2:

> [..] If T is a physical type or subtype, the result is expressed in terms of the primary unit of T unless the base type of T is TIME, in which case the result is expressed in terms of the resolution limit (see 5.2.4.2); in either case, if the unit is a basic identifier, **the image of the unit is expressed in lowercase characters**.

> Then this issue is discussed and solved, right?

Yes I think the error about 9223372036854775808 is allowed by the LRM since it's outside the range the `universal_integer` is required to support, and `freq'image(5 Hz)` should evaluate to `"5 hz"`.