nickg/nvc #1286: Mixed language crash [Verilog + VHDL] - axis2mm

mik1234mc, 2025-09-04:
Hello @nickg ,

thank you for massive updates, I prepared a small tb around axis2mm.v to check the compiler functionality. I attach it. Just run `make` and you should see the following output. The issue is in file axis2mm_tb.vhdl line 239. If you comment out `m_axis_tvalid <= '1';`, the error disappear. If fixed, I can continue with tb implementation.

Regards,
Michael

[nvc_test_compile.zip](https://github.com/user-attachments/files/22132283/nvc_test_compile.zip)

```
nvc -a ./sfifo.v
nvc -a ./skidbuffer.v
nvc -a ./axis2mm.v
nvc -a ./axi_pkg.vhdl
nvc -a ./axis2mm_tb.vhdl
nvc -M 512m -e axis2mm_tb
nvc -M 512m -r axis2mm_tb --wave=tmp.ghw
** Note: 200001ps+0: AXI-Lite Write: Setting AXI4 Base Address
   Procedure AXILWRITE [AXIL_O_T, AXIL_I_T, STD_LOGIC_VECTOR, STD_LOGIC_VECTOR] at ./axi_pkg.vhdl:45
   Process :axis2mm_tb:_p12 at ./axis2mm_tb.vhdl:219
** Note: 225001ps+0: AXI-Lite Write: Setting AXI4 Base Address
   Procedure AXILWRITE [AXIL_O_T, AXIL_I_T, STD_LOGIC_VECTOR, STD_LOGIC_VECTOR] at ./axi_pkg.vhdl:45
   Process :axis2mm_tb:_p12 at ./axis2mm_tb.vhdl:224
** Note: 255001ps+0: AXI-Lite Write: Setting AXI4 Base Address
   Procedure AXILWRITE [AXIL_O_T, AXIL_I_T, STD_LOGIC_VECTOR, STD_LOGIC_VECTOR] at ./axi_pkg.vhdl:45
   Process :axis2mm_tb:_p12 at ./axis2mm_tb.vhdl:229
** Note: 285ns+2: Sending AXI-Stream data
   Process :axis2mm_tb:_p12 at ./axis2mm_tb.vhdl:200

*** Caught signal 11 (SEGV_MAPERR) [address=(nil), ip=0x55d1ec6c6f29] ***

[0x55d1ec59f225] ../src/util.c:906 signal_handler.lto_priv.0
[0x7fb34649c13f] (/usr/lib/x86_64-linux-gnu/libpthread-2.31.so) 
[0x55d1ec6c6f29] ../src/rt/copy.c:1305 cmp_bytes_sse41
[0x55d1ec6c6f29] ../src/rt/copy.c:101 _cmp_bytes
[0x55d1ec6c6f29] /usr/lib/gcc/x86_64-linux-gnu/10/include/emmintrin.h:97 _cmp_bytes
    }
    /* Create an undefined vector.  */
[0x55d1ec6bf7b3] ../src/rt/copy.h:45 sched_driver.lto_priv.0
[0x55d1ec6847ac] ../src/rt/model.c:4134 x_sched_waveform
[0x55d1ec6847ac] ../src/rt/model.c:295 __nvc_sched_waveform
[0x55d1ec6857bd] ../src/jit/jit-exits.c:486 __nvc_do_exit
[0x55d1ec688aa4] ../src/jit/jit-interp.c:844 interp_loop
[0x55d1ec688aa4] ../src/jit/jit-interp.c:1146 jit_interp
[0x55d1ec66fdd0] ../src/jit/jit-core.c:700 jit_try_vcall.lto_priv.0
[0x55d1ec670410] ../src/jit/jit-core.c:799 jit_fastcall
[0x55d1ec6be3d2] ../src/rt/model.c:984 run_process
[0x55d1ec6be3d2] ../src/rt/model.c:2739 async_run_process.lto_priv.0
[0x55d1ec6bc3b3] ../src/rt/model.c:407 deferq_run.lto_priv.0
[0x55d1ec5989c7] ../src/rt/model.c:3409 model_run
[0x55d1ec5989c7] ../src/rt/model.c:1019 run_cmd
[0x55d1ec59a498] ../src/nvc.c:2458 process_command
[0x55d1ec593de1] ../src/nvc.c:2620 main

nvc 1.18-devel (1.17.0.r109.g39af564d) (Using LLVM 11.0.1) [x86_64-pc-linux-gnu]

Please report this bug at https://github.com/nickg/nvc/issues

make: *** [Makefile:8: all] Error 2
```

nickg, 2025-09-06:
I fixed that crash and it gets a bit further. Now the assert on line 268 is failing:

```vhdl
assert m_axi_awlen  = std_logic_vector(to_unsigned(num_words-1, 8)) report "AXI4 Length mismatch" severity failure;
```

I guess it should be `to_signed` to handle the negative argument (or perhaps `(num_words - 1) mod 256`?) but `m_axi_awlen` is `X"0f"` here. I haven't investigated this yet.

nickg, 2025-09-20:
@mik1234mc I think this testbench is racy because it samples and updates signals in the same cycle that the UUT does so the behaviour depends on how VHDL and Verilog delta cycles are interleaved which I don't think is specified anywhere. If I restructure the main loop to sample/update on the falling edge of the clock then the test passes in Questa and NVC (latest master only as there were some other bugs):

```vhdl
    wait until falling_edge(clk);

    -- Check data
    for i in 0 to DATA_SIZE-1 loop
      while (m_axi_wvalid /= '1') loop
          wait until falling_edge(clk);
      end loop;
      m_axi_wready <= '1';
      report "Received data word " & integer'image(i) & " at AXI4: " & slv2bstr(m_axi_wdata) severity note;
      assert m_axi_wdata = test_data(i) report "Data word " & integer'image(i) & " mismatch" severity error;
      if i = DATA_SIZE-1 then
        assert m_axi_wlast = '1' report "tlast signal not asserted" severity error;
        report "AXI4 tlast signal OK" severity note;
      else
        assert m_axi_wlast = '0' report "tlast signal asserted early" severity error;
      end if;

      wait until falling_edge(clk);
    end loop;
```

mik1234mc, 2025-09-22:
Thank you for upgrades. I expected delta cycle problems between verilog and vhdl. It will be interesting to check it with synthesizable vhdl rtl instead of this testbench to check if it can run from the same clock edge.

nickg, 2025-09-28:
The scheduling differences were caused by the clock gating inside the Verilog code. I've adjusted the way continuous assignments are scheduled so it works like Modelsim and the original testbench now passes.

closed 2025-09-28 by nickg